ArduPilot/ardupilot #1924: APMrover - init problem for Pixhawk - double yellow blinking

mw46d, 2015-02-27:
The commit *0a68d4ef39952dca08f3040e74208d727510906a* for libraries/AP_Notify/AP_Notify.cpp exposed a problem for the APMrover where a couple of flags are set before the updated AP_Notify::init() method. With the above commit, the AP_Notify::init() method now resets those flags.

The proposed fix is to move the AP_Notify::init() above the setting of those flags:
```
diff --git a/APMrover2/APMrover2.pde b/APMrover2/APMrover2.pde
index 40f1894..75235be 100644
--- a/APMrover2/APMrover2.pde
+++ b/APMrover2/APMrover2.pde
@@ -537,13 +537,14 @@ void setup() {
     // load the default values of variables listed in var_info[]
     AP_Param::setup_sketch_defaults();
 
+    notify.init(false);
+
     // rover does not use arming nor pre-arm checks
     AP_Notify::flags.armed = true;
     AP_Notify::flags.pre_arm_check = true;
     AP_Notify::flags.pre_arm_gps_check = true;
     AP_Notify::flags.failsafe_battery = false;
 
-    notify.init(false);

    rssi_analog_source = hal.analogin->channel(ANALOG_INPUT_NONE);
 
```

gmorph, 2015-08-07:
This issue was resolved in a previous Rover release. 
Thanks!

closed 2015-08-07 by gmorph

magicrub, 2015-08-07:
But is it fixed now?

gmorph, 2015-08-07:
Yep its fixed.