ArduPilot/ardupilot #1926: APMrover 2.48 @Pixhawk - Telemetry2/Serial2 does not seem to work:-(

mw46d, 2015-02-27:
It works fine for 2.47 (did a downgrade to verify), so my wiring and everything is ok. But as soon as I run 2.48 (or current HEAD), the Pixhawk does not seem to send anything out via the Telemetry2 port:-( 

The two images show the traffic on the Telemetry2/Serial2 lines, the 247 is the working example (Heartbeat from the Odroid in Digital0, Status messages from the Pixhawk in Digital1)
[image]

Only the Heartbeat shows up for the 2.48 (release) in this case:

[image]

mw46d, 2015-02-28:
This might be related to #1854 . After a couple of resets and down/upgrades (between 2.47 & 2.48) of the firmware it seems to work for now. But it worked in between as well for some time and then stopped again for no apparent reason:-( So there is something strange going on:-(

khancyr, 2017-08-25:
Closed, it works on pixhawk !

closed 2017-08-25 by khancyr