ArduPilot/ardupilot #1928: Plane: DO_CHANGE_SPEED as first WP causes mission bug

Georacer, 2015-02-27:
Conditions under which the bug occured:
- The first mission item was a DO_CHANGE_SPEED (nav_cmd no 178) (with airspeed value 33).
- The second mission item was a TAKEOFF command.
- Other DO_CHANGE_SPEED commands occured in the mission (with values other than 33).

Every time a new location waypoint (nav_cmd 16) was becoming the current one, a DO_CHANGE_SPEED command was also executed, with the value of the first mission item (WP1).

Removing the first mission item (making TAKOFF item no1) clears the bug.

Georacer, 2015-02-27:
Note: If a DO_CHANGE_SPEED mission item was placed after a Waypoint mission item, then the bugged DO_CHANGE_SPEED command would not appear.

omi10859, 2018-08-14:
>A log file can be found here: http://ardupilot.com/forum/viewtopic.php?f=44&t=11426&sid=4edd9ca14d71409cbdafde3bd51064f0

link is broken

peterbarker, 2018-08-14:
@omi10859 Are you seeing this bug as still present?  This is a pretty old issue now, I'm *hoping* the resolution is "someone fixed it in the past" :-)

That being said - @Georacer 's description was pretty solid, and it should be reproducible in SITL.

Thanks for chasing this up a little!

omi10859, 2018-08-16:
I'm not sure that bug is still there or not
Was just searching some good first issues to start contributing and landed here :)

KimHyungSub, 2021-05-31:
I tested this bug on 0bb18a153c and it seems the bug is fixed.

peterbarker, 2021-06-01:
Thanks @KimHyungSub - closing this

closed 2021-06-01 by peterbarker