ArduPilot/ardupilot #1934: Antenna Tracker: DO_SET_SERVO missing

SCUBA82, 2015-03-02:
There is an option to test the pitch/yaw servos in Mission Planner. It sends the "do_set_servo"  command to the antenna tracker. Unfortunately this command isn't implemented in the current antenna tracker version (v0.4). I would recommend to implement the option in antenna tracker or remove it from mission planner. 

I'v played around with rmackay9's "antenna-wipe6" branch which has this command implemented but there seems to be some issues. Pressing the "yaw test" button moves the pitch servo, pressing the "pitch test" button has no action. I also checked the "ch1(2)_out" values to be sure there is no hardware error.

rmackay9, 2015-03-02:
@SCUBA82, yes, we should certainly add this as it's quiet close to completion in that branch you saw and it would help people verify their setup is ok.  thanks for the reminder.

SCUBA82, 2015-03-02:
@rmackay9 thanks for your quick reply.Yep, and it's very confusing when there is an option that doesn't work and is not completely documented.

I think i've fixed the issue with the "do_set_servo" routine in your branch: Mission Planner sends the parameter "1" for YAW and "2" for PITCH. But you're checking i.e "if (servo_num == CH_PITCH)" where "CH_PITCH" is definded with "CH_2" and not "2".

KtotheG, 2015-03-05:
+1. This seems to be the only way to verify your numbers if using an APM as the control board

rmackay9, 2015-04-27:
This should be fixed with this commit and should appear in the MP in an hour or so as ver 0.5.
https://github.com/diydrones/ardupilot/commit/f74d2101d173d0f3a3d76b4aab5c125db0b85093

Sorry for the delay!

closed 2015-04-27 by rmackay9