ArduPilot/ardupilot #1935: pre-arm mode not armable [PIXHAWK]

bl4ckc00k1e, 2015-03-02:
What that means?? sometimes i have the High GPS HDOP but in a few minuts disapear, but later i have this message:
"pre-arm mode not armable"

If i Disable the precheck, i have the same issue...

I am searching on google, but nothing... I have arducopter 3.2.1
PLEASE HELP

rmackay9, 2015-03-02:
@bl4ckc00k1e, that usually happens when you're trying to arm in a mode that you're not allowed to arm in like LAND or AUTO.

closed 2015-03-02 by rmackay9

bl4ckc00k1e, 2015-03-02:
OMG its true, thanks very much!

epullah, 2016-05-14:
how to fix arm:mode not armable if I set to do an autopilot.

i am using apm2.6..Please help me

magicrub, 2016-05-14:
You've posted a how-to help question in the developers bug tracking system,
that's the wrong place. Please repost in the forums at discuss.ardupilot.org

suraj2596, 2019-12-01:
I'm arming it in GUIDED mode and yet it the motors wont move. I get `taking off` but nothing happens

rmackay9, 2019-12-02:
@suraj2596, this sounds like a support question so I think it needs to be raised in the [Copter-3.6](https://discuss.ardupilot.org/c/arducopter/copter-36) or [Copter-4.0 forum](https://discuss.ardupilot.org/c/arducopter/copter-40).  Sorry to be strict but otherwise our bug tracking/to-do list turns into a support forum.  Thanks for using AP!

suraj2596, 2019-12-02:
@rmackay9 My apologies. I actually figured it out. It's the GPS lock, sometimes it works and sometimes it doesn't. Just a thought, what if the firmware can tell the users which pre-arm check exactly failed rather than outputting ack result 4. It's hard to debug. Good day :)

rmackay9, 2019-12-02:
@suraj2596, yes, we've discussed adding a mavlink message that would provide the information in a machine-readable format.  For now we send text messages which start with "PreArm:" or "Arm:"

suraj2596, 2019-12-02:
Oh, that's great!