ArduPilot/ardupilot #1941: ’feenableexcept’ doesn’t exist on OSX.

davidbuzz, 2015-03-03:
please cherry-pick this tiny fix: 
https://github.com/davidbuzz/ardupilot-mega-buzz/commit/bf8dc3aa35570a2671ff67fda6cd849fad919b68

davidbuzz, 2015-03-03:
try compiling diydrones/master on OSX, and this error occurs: 
‘make sitl’ gives me this: “/Users/buzz/UAV/ardupilot/libraries/AP_HAL_AVR_SITL/Scheduler.cpp:213:9: error: use of undeclared identifier 'feenableexcept'; did you mean 'feraiseexcept'?"
the above fixes it.

rmackay9, 2015-04-14:
I'll bet doing this as a PR will make it more likely that it gets into master.

OXINARF, 2017-06-02:
This has been fixed. Although AVR doesn't exist anymore, current SITL also uses feenableexcept. A workaround for OSX was included: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Common/missing/fenv.h

closed 2017-06-02 by OXINARF