ArduPilot/ardupilot #1943: request_data_stream rate misinterprets a frequency for a period

phelpsw, 2015-03-05:
Specifically  [here](https://github.com/diydrones/ardupilot/blob/2aae59437136fe610f8b899ac3fbd2111f3694b4/libraries/GCS_MAVLink/GCS_Common.cpp#L467) ardupilot misinterprets the mavlink spec by implementing the message rate as a frequency rather than a period.  It is clear in the [mavlink documentation](https://github.com/mavlink/mavlink/blob/b160f0e3c7174c115117eece93be691488faaf51/message_definitions/v1.0/common.xml#L1929) that a period was intended.  To quote the description of the parameter req_message_rate, "The requested interval between two messages of this type"

I'm a proponent of strictly following the mavlink spec in this instance (treat these rates as periods) as it makes it easier to specify not only *high* frequency messages (the typical 1-10Hz sort of thing), but also *low* frequency messages where we only care about an update once every minute or so.  This sort of thing is useful for long range, low bitrate communications systems (ie satellite links etc).

rmackay9, 2015-03-05:
I've checked with Lorenz and apparently you're right!

HumamHelfawiOptecks, 2017-06-27:
is it currently a frequency or period?

OXINARF, 2017-06-28:
It is still frequency. I should add that the MAVLink spec was changed and it no longer mentions it is a requested interval. Also, other autopilots and the GCSs also use it as frequency.
I'll say it is established that it should be frequency, so I'm closing this.

closed 2017-06-28 by OXINARF