ArduPilot/ardupilot #1944: Mode warning

abvgeej, 2015-03-05:
It should be possible to either forbid arming in Sport/Acro modes, or to have a special warning when arming.

rmackay9, 2015-03-05:
My view is that this is really a GCS responsibility.

JoshWelsh, 2015-03-05:
I would agree with that, since we're talking about arming from the GCS I hope and not the RC transmitter.. The title concerned me initially until I saw Randy's reply :)

abvgeej, 2015-03-06:
@JoshWelsh No, i am talking about RC.
We should care about the users who have memory problems, and dont need to arm in Acro :)
Of course, there are users which want to arm in those modes, but i would prefer to at least have some beeping and blinking on me before i do something i dont do that often.

R-Lefebvre, 2015-03-06:
What exactly is the issue you're trying to solve here?

abvgeej, 2015-03-06:
1) Have a different beep pattern or LED warning when arming in those modes.
2) Get a option to forbit arming in those modes.
(As i know, you cant arm in Drift/Circle, but if i am wrong, that also applies to them)

R-Lefebvre, 2015-03-06:
I'm asking why you would want this?

abvgeej, 2015-03-06:
@rmackay9 "We should care about the users who have memory problems"
Mode switches are a thing you can always forget about.

lthall, 2015-03-06:
Is the intention of this is to warn people that they haven't checked what mode they are in before arming the copter?

If you forget about your mode switch then you forget about safety and shouldn't be flying.

I am speechless!!!

R-Lefebvre, 2015-03-06:
I think you should just remove acro and stabilize modes from your switches?

abvgeej, 2015-03-06:
@lthall Well, this is still a good safety feature. I fly well, but constantly forget the antennas at home :)
After all, we have arm locks on LAND, RTL and many more modes. Why cant we have a optional one on those?

@R-Lefebvre But i still wish to use them. (And what does "Stabilize" have to do with this?)

I start to realize how utterly stupid this feature sounds, and all of my comments on it. Close this, as I feel ashamed.

JoshWelsh, 2015-03-06:
@abvgeej since you opened it you can close it yourself, you don't have to ask Randy or the others to close it.
I think Stupid is too strong a word, as your intention sounds decent enough, but I echo Leonard's thoughts as well as Robs that this is a setup and safety awareness issue. I'm glad you've come around on this request now!

abvgeej, 2015-03-06:
@JoshWelsh As i know, i dont have the rights to close this issue, but if you know a way i can, it would be great.

rmackay9, 2015-03-07:
the overall goal of trying to improve safety is a worthy goal.
I'll close this as agreed above.

closed 2015-03-07 by rmackay9