ArduPilot/ardupilot #1946: Tracker pitch issue

KtotheG, 2015-03-05:
Tracker only pitches when in scan mode. In auto mode the pitch holds steady around the midpoint. This is with a pixhawk on the plane, apm running tracker v.04, 3dr radio connected to each, and the tracker connected to mission planner via usb. Looking at the tuning you can see that the apm shows desired pitch changes but actual pitch does not. 
[image]
[image]

KtotheG, 2015-04-13:
Bump!!! Could somebody look into this issue please

KtotheG, 2015-04-13:
Bump!!!! Could somebody look into this please

KtotheG, 2015-04-23:
Randy, the modifications made to the code seems to have fixed the pitch issues. Thank you for taking the time to make that correct. But since then, when trying to connect to the plane through telemetry, it takes more then an hour to load the planes parameters. The apm 2.6 is using uart2. There is no parameter setting to change the uart2 speed. Is this an addition that could be added to the code or is there another fix? As always, thanks for all your hard work.

rmackay9, 2015-04-27:
Re the download speed issue by doing this:
1. connect 3DR radio to GCS Windows machine set to NetID 24
2. connected 3DR radio to Tracker's Pixhawk on Telem2 using NetID 24
3. connected another 3DR radio to Tracker's Pixhawk on Telem1 using NetID 25
4. connected 3DR radio to Vehicle's pixhawk on Telem1 using NetId25
With this setup I was able to download parameters from the vehicle without any problems.  It took about a minute and a half.

I know this is not exactly a perfect test because I used Pixhawks instead of APM2s so the boards I used had a lot more CPU power and RAM.

KtotheG, 2015-04-27:
I don't have 2 pairs of radios to test this method. I have the Apm on the plane and pixhawk on the tracker. I modified the apm to originally work on the tracker so the radio is on uart2. The pixhawk, on the tracker, has the radio on telemetry 1 and usb connected to gcs. Regardless of the configuration, if uart2 is used on an apm, shouldn't there be parameters to configure it properly. I can't change anything on uart2 other then baud rate and if it is sending mavlink, frisky, etc. I could be completely wrong but it seems like this is why the plane parameters take 2 hours to load.

KtotheG, 2015-05-07:
You have since fixed this issue randy. I will close it.

KtotheG, 2015-10-07:
I uploaded the newest version of the firmware and the pitch problem continues. I haven't used it for a while so I can't tell you exactly when the problem started. By that I mean which version. The pitch just sits at the midpoint and doesn't track. Hoping we can find a solution to this soon. This has been a very long project

Gabriele1995, 2015-12-07:
I have a similar problem with tracker v0.7.3 on pixhawk connected to 3DR GPS, 3DR Radio v2, and two servos (1 and 2 outputs). Yaw works and follows the plane perfectly, but pitch doesn't move in auto mode at all. In scan mode it works properly, but in auto it just keeps pointing at 0 and doesn't move, no matter how high/low the plane is.

rmackay9, 2015-12-27:
I've released 0.7.4 (and more recently 0.7.5) which I think includes a fix to this issue.  I believe the issue was the the tracker was not asking the vehicle to send it's baro pressure and it seems it's this pressure that is used (after being converted to an altitude) to calculate the pitch.

I've tested and confirmed that the pitch on my tracker at least is working.

I've also added dataflash logging so if anyone is still suffering from this problem with pitch not working, can you update to 0.7.5 perform a test and then send me a dataflash log?  You can provide the link to the log here or in the "figuring out the antenna tracker" discussion on diydrones.com.  Another possible place is in the APM:Forum but I don't check that much.

closed 2015-12-27 by rmackay9