ArduPilot/ardupilot #1948: PID Banks

MallyAzure, 2015-03-05:
I've often come in situations that require a different PID setting like changing battery to a different cell count, adding gimbal and sometimes just wanting a more softer response. Implementation of PID banks (also maybe assignable to 3 way switch on ch 6/7/8) would make things alot easier than having to manually upload a different parameter list from a GSC each time you want to change.

KIrill-ka, 2015-03-05:
I second this. A larger battery or additional camera would increase weight and require change of PID. It would be handy to have a number of preconfigured sets.

JSGordon, 2015-03-14:
I have also second but think we should look at all features that should be included in a mode bank.  For instance in the situation quoted above where weight is changed due to payload change I have found the throttle mid point to be an issue.  If you remove significant weight and switch from alt hold to manual the copter will rocket up or if increasing load will drop.  There may be other features that would be useful as well.  IMHO I would widen the issue to "mode bank".

MallyAzure, 2015-03-14:
Yes, didn't think of that, totally agree with you, mode banks for different setups and configurations.

sreid55, 2015-04-16:
I suggest we call these "Profiles"; they could include PIDs, battery capacity (mAH), and other selected parameters.   Obviously a big project on both the firmware and ground station side, but really useful.
Actually, now that i think about this, there's no reason this couldn't be fully / only implemented in the ground station software.

Naterater, 2019-01-05:
I think this should be handled by ground stations rather than the autopilot because it is a 1-time change on the ground before the flight. I also like the feature, but I think it belongs here: https://github.com/ArduPilot/MissionPlanner/issues.

IamPete1, 2021-08-08:
This can be done with scripting.

auturgy, 2022-01-05:
As Pete says above, this can now be achieved via a Lua script.  Closing.

closed 2022-01-05 by auturgy