ArduPilot/ardupilot #1950: Feature: Battery EMF monitoring instead of voltage

msk7-ripe, 2015-03-05:
As I understand (correct me if I'm not right) LiPo cell has 4.2V EMF fully charged and 3.0V EMF fully discharged. Now AP monitors battery voltage, not EMF. These show same values when there's no load. When under load E=ur + UR, where "E" stands for EMF, "u" is voltage drop on battery internal resistance "r" and "U" is voltage measured on load resistance "R". Now AP shows "U" and this value changes under different load. This prevents accurately estimating battery charge level by voltage. My idea is to show "E" instead of "U". When AP is first powered (battery is connected) there is almost no load and we can measure initial EMF and store it as "E". When AP is armed and current goes to some reasonable value for measuring battery resistance (e.g. 10-20A), AP calculates "r" once as r=(E-U)/I. After that AP uses "r" value to calculate EMF as E=U+Ir and shows this value instead of "U". This will allow pilot to see real battery state, use it's full capacity when needed and avoid over-discharge. 

Yes, I understand that internal resistance will grow slightly through battery discharge, but I think method suggested is accurate enough.

magicrub, 2015-03-05:
There has been talk about logging, on a per-boot basis, the internal
resistance of the battery in a similar manner in an attempt to detect
degraded battery health. Some of that discussion would overlap here.

rmackay9, 2015-03-06:
By the way, there is an early attempt at measuring the battery resistance in master here:
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_Motors/AP_Motors_Class.cpp#L300

msk7-ripe, 2015-03-06:
I expected I invented nothing new, but did my search in batt monitor lib, not motors.

Naterater, 2019-01-05:
I know this is already implemented in Plane as "sag-compensated voltage." I believe it is in copter too. If not, please re-open.

closed 2019-01-05 by Naterater

rmackay9, 2019-01-07:
Yup, it's in Copter as well