ArduPilot/ardupilot #1951: Copter: Flight safety Interlock needed for Copter in flight.(Mode Stab while thr=0)

hotelzululima, 2015-03-07:
while the "official" position of craig elder and possibly others is allowing a mode change in APM from an RC TX to STAB  with THR = 0 is a reasonable request  for APM.

 I submit it is an OPEN malfunction and flight safety issue.

while this request is reasonable for a plane autopilot ,

 allowing in Arducopter APMs mode to be reset  to STAB while the RC throttle stick position
  is not equal to or greater than the current flight throttle setting is a definite FAIL/BUG and results in a crash when combined with  RC/TX radio errors while running from a tablet/telemetry as primary GCS.

hotelzululima, 2015-03-07:
whoops.. this is for ArduCopter Release 3.2.1  :)
     hzl

hotelzululima, 2015-03-07:
its occurs that the mode change to stabilize when commanded CAN safely be allowed in arducopter  if throttle stick position is at hover throttle or greater...
   hzl

R-Lefebvre, 2015-03-07:
So you're suggesting that, if somebody is flying in some mode, such as Loiter, Alt_Hold, etc and they are descending, and something goes wrong and they need to stop the propellers, that they should be required to switch to Stabilize mode, and then RAISE the throttle and then lower it, would improve the safety of the system?

hotelzululima, 2015-03-07:
yes I am.. by not having unmanned copters rain down in innocent 3rd parties in the parks we fly in...

   hzl

lthall, 2015-03-07:
This is why all regulatory bodies and model association bodies prohibit flying over people!

Copied from another issue:

The current assumption Arducopter works on is that the RC transmitter is the primary control and all other methods of control are secondary. We could implement a ch7 or ch8 switch that is used to define that we are not operating the RC transmitter and are instead operating from the GCS. This could then be used to ensure we enter Loiter, RTL or Land (or what ever we think is appropriate) if we loose GCS (or switch back from the GCS). The RC transmitter operator can then take control at any time by flicking that Ch7/8 switch back to RC control.

rmackay9, 2015-03-09:
I think this is a duplicate of the issue raised by Marco here: https://github.com/diydrones/ardupilot/issues/1797

closed 2015-03-09 by rmackay9

hotelzululima, 2015-03-09:
actually this is a variant of the bug hit by marco and doesnt follow  the same path through the code.. thus its been PROPERLY filed as a separate bug .
 lthall has contributed nothing but noise about the above comment..

this relates to TABLET GCS operation of IRIS/IRIS+  with APM 3.2.1  Marcos issue was about NONE GCS operation this is about  GCS  MODE with a tablet NOT an RCTX in use..

This issue was filed for compliance with California Product Safety law to Inform 3DR the patron of these groups that their product is defective according to California Product Safety Law.

These issue WILL be filed in spite of you Randy McKay closing them in a negligent fashion.. none of this will bode well in court given the actions I have seen here.. Rob has been the only one to address this safety defect in a reasonable and responsible manner.

I suggest you actually reopen the issue  and take a look at the code like I have  been doing to trace this behavior.. BTW documentation of a dangerous behavior of a product does NOT release one(3DR) from liability in a California court from 3rd party liability.

rmackay9, 2015-03-09:
@hotelzululima, I've deleted your comment because of the personal comments about lthall.  Feel free to repost without that line.

I'm going to delete any other responses from anyone else as well if they make personal comments about anyone else on this thread.

hotelzululima, 2015-03-09:
uh then inform lthall not to attack me personally in the first place..!

hotelzululima, 2015-03-09:
and  I have saved all these comments off and will be offering them during any litigation support on any product liability safety case involving 3DR in california.. matter of law you see..

hotelzululima, 2015-03-09:
actually this is a variant of the bug hit by marco and doesnt follow the same path through the code.. thus its been PROPERLY filed as a separate bug .

this relates to TABLET GCS operation of IRIS/IRIS+ with APM 3.2.1 Marcos issue was about NONE GCS operation this is about GCS MODE with a tablet NOT an RCTX in use..

This issue was filed for compliance with California Product Safety law to Inform 3DR the patron of these groups that their product is defective according to California Product Safety Law.

These issue WILL be filed in spite of you Randy McKay closing them in a negligent fashion.. none of this will bode well in court given the actions I have seen here.. Rob has been the only one to address this safety defect in a reasonable and responsible manner.

I suggest you actually reopen the issue and take a look at the code like I have been doing to trace this behavior.. BTW documentation of a dangerous behavior of a product does NOT release one(3DR) from liability in a California court from 3rd party liability.

hotelzululima, 2015-03-09:
actually this is a variant of the bug hit by marco and doesnt follow the same path through the code.. thus its been PROPERLY filed as a separate bug .

this relates to TABLET GCS operation of IRIS/IRIS+ with APM 3.2.1 Marcos issue was about NONE GCS operation this is about GCS MODE with a tablet NOT an RCTX in use..

This issue was filed for compliance with California Product Safety law to Inform 3DR the patron of these groups that their product is defective according to California Product Safety Law.

these issue WILL be filed in spite of anyone closing them in a negligent fashion.. none of this will bode well in court given the actions I have seen here.. Rob has been the only one to address this safety defect in a reasonable and responsible manner.

I suggest you actually reopen the issue and take a look at the code like I have been doing to trace this behavior.. BTW documentation of a dangerous behavior of a product does NOT release one(3DR) from liability in a California court from 3rd party liability.