ArduPilot/ardupilot #1953: Copter: ENHANCEMENT flight replay from logging file but be able to change flight parameters to see what if

hotelzululima, 2015-03-09:
Hi All,
       this may be in a FAQ or the wiki.. I did  not see it so will ask if this can be a possibility.

Is it possible to use the bin log to simulate a flight in toto(similar to SITL) while changing parameters such as failsafes in various parts of the flight  to see if the flight has a different ending and what that would be.
Its would seem that this would not be impossible given the components and data to work with..
I am thinking about something similar to exmaps.com  but with the ability to alter flight parameters in the FC..

Any interest in this type of feature set.. or did someone already do this

robustini, 2015-03-09:
I like it...

hotelzululima, 2015-03-09:
can it be done with tlog already Marco?  any urls?

     hzl

hotelzululima, 2015-03-10:
I tried modeling my issue on http://www.exmaps.com/uploads/11257  but of course no way to do whatif modeling with the loaded parameters failsafes and PIDS, I suspect we can get farther by using SITL to setup the simulation  checking on that now...

hotelzululima, 2015-03-12:
Hi marco..
 found this http://dev.ardupilot.com/wiki/simulation-2/sitl-simulator-software-in-the-loop/using-sitl-for-ardupilot-testing/  but they DON'T have a facility for rc_transient errors looking at code now to see how hard to simulate that  and making sure there is a test case for same...

     hzl

robustini, 2015-03-12:
No, that's another thing.
It would be very interesting if the Planner could load the logs as it does with the tlog (Play, Pause, etc.).

hotelzululima, 2015-03-12:
looks like this may be it though http://dev.ardupilot.com/wiki/simulation-2/testing-with-replay/

hotelzululima, 2015-03-12:
just need to make sure I can 3.2.1 release.. and wondering how it will sim the rcrf transient error from the bin log.. 

also need to see if the  FS was triggered even if set to do nothing is logged for what I am looking for..
     hzl

hotelzululima, 2015-03-12:
replay does not look like it either... some combo of it an SITL perhaps

peterbarker, 2021-04-15:
Modern Replay allows you to rerun sensor data through the EKF estimators with different parameter values to see how it would have performed.

Our normal SITL environments allow for re-running of mission files, and addition of a physics backend of suitable fidelity allows for reasonable flight reproduction which could be used for testing different parameters.

Pretty sure there's nothing in-between.

Closing this now - it's very old and the ecosystem has evolved a rather large amount making the question out-of-date.  Its really more of a forum question anyway :-)

closed 2021-04-15 by peterbarker

robustini, 2021-04-16:
https://plot.ardupilot.org/#/