ArduPilot/ardupilot #1954: Camera trigger using "relay" overrides RELAY_DEFAULT

mareschn, 2015-03-09:
When setting RELAY_DEFAULT parameter to 1, the setting is overridden when also setting the CAM_TRIGG_TYPE parameter to 1...  I'm trying to trigger on low logic, rather than high.

(setting RELAY_DEFAULT to 1 should make the aux pins default high, setting the cam_trigg_type to 1 should then cause the output for aux 5 to go low when the camera is triggered) Tested on arducopter and arduplane.

Would it be hard to enable RELAY_DEFAULT to function in conjunction with the CAM_TRIGG_TYPE? In addition to avoid problems with other aux pins, it might be nice to have separate RELAY_DEFAULT parameters for each aux pin.

Naterater, 2019-01-05:
I think some of the logic here has changed since the removal of NuttX. Can you recreate the issue on a new ChibiOS build?

WickedShell, 2019-01-06:
No this is a problem that the camera library will only trigger the first relay and ignores the `RELAY_DEFAULT` option.

WickedShell, 2019-05-28:
Reopening as this hasn't actually been resolved.