ArduPilot/ardupilot #1961: Add boolean to indicate whether stall prevention is active

phelpsw, 2015-03-10:
As discussed here: http://ardupilot.com/forum/viewtopic.php?f=107&t=11505 it would be useful to have a bit which indicates whether the stall prevention code is active.  If stall prevention is excessively active, it could confusingly prevent a mission from being completed as expected.  This bit will make it clear in telemetry that stall prevention prioritizing keeping the aircraft aloft over directly following commands.

magicrub, 2015-06-10:
I don't see the need for a SYS_STATUS bit as @tridge suggested. This data is available by knowing what mode you're in and the STALL_PREVENTION param which are both visible to the GCS and in the logs.
http://plane.ardupilot.com/wiki/arduplane-parameters/#enable_stall_prevention_arduplanestall_prevention

@Tridge, if you think this is still necessary I'll leave it open but otherwise I'd say lets close it.

magicrub, 2016-06-06:
there is now a TECS.f "flag" variable where the underspeed flag state is available. Flag is defined here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_TECS/AP_TECS.h#L235 so this helps in after-flight dataflash review and not in real-time flight. I'm closing this for now, if we really need a mavlink based status we/you can re-open.

closed 2016-06-06 by magicrub