ArduPilot/ardupilot #1966: Logging GPS raw observables data on pixhawk

Naish88, 2015-03-11:
A very useful enachement for surveying application, is the ability to post process the GPS raw observables to obtain geotagging of images with cm accuracy. 

This can be accomplished in 3 ways on pixhwak with an Ublox M8N:

GPS Serial port 1 receives a mix of Ublox protocol data and RAW data, a simple parsing of data can separate the "mix". 

GPS serial port 1 receives the Ublox protocol, and the correction data is streamed on the I2C bus form the Ublox module. 

Use of 2 ublox receivers, one goes in Gps port 1 ( with Ublox protocol ) and the second one goes in Gps port 2 ( with Raw data) 

All the Raw data needs to be saved on the SDcard.

krolladam, 2015-03-15:
I was speaking with a photogrammetry company that have the best UAV setup I've ever seen and this is what they do.  They record the raw data on the UAV as well as on the ground and post process using rtk lib, rather then doing rtk on the fly.  If they lose comms with the ground station using the traditional method they need to refly that part of the flight, but by post processing it, they don't need to refly any of their missions.  This is definitely a good move.

cartoocm, 2015-03-16:
Hi. This is my first post :)
Simple schema for one solution

[image]
Regards

tridge, 2015-03-16:
I agree that this is very worthwhile to add. We'll just need to be careful about running the GPS UART at a high enough speed to get the data reliably

cartoocm, 2015-03-16:
Hola Andrew, I'm happy to talk to you. I just sent to my mate Dario (aka @taedla01 ) a possible line of work. My experience as a programmer is very short, but I'm an engineer in geodesy and photogrammetry. Since this time I hope to actively collaborate in the project.
I think the bottleneck may be, storing the raw data in microSD, with a recommended minimum frequency of 2 Hz. Also, I recommend 2 receivers GNSS raw data logging for redundancy.
@taedla01 tomorrow will send you a small brainstorming. Sorry for my bad English.

Naish88, 2015-03-16:
@cartoocm : Raw data at 2hz won't use much space, and there is plenty of space on the SD card.
Won't be easy to log raw data form 2 GPS units, as pixhawk has only 2 GPS com ports and one must be used for GPS position data ( uBlox protocol ). Plus having redundancy on GPS raw is a bit useless if you have no redundancy on camera, sd card and other things...

WickedShell, 2015-07-20:
This has been completed for ublox receivers with RAW or RAWX data support. Theres only one pull request pending #2562 that brings the increased baud rate, and fixs a bug with having more then 15 satellites, but otherwise support is in, and behaves much as described in option 1, except that all data ends up in the same APM log file.

If there are any other interesting messages that need to be logged then please reopen this with what messages are needed.

closed 2015-07-20 by WickedShell