ArduPilot/ardupilot #1968: Display all channel outputs in MP?

GrahamDyer, 2015-03-11:
MP Issue #812 refers

The Pixhawk supports outputs to more than 8 channels but ever since APM2.x days the MP has always only shown ch 1-8 outputs. Please can the outputs to the additional channels be shown in Status?
[image]

JSGordon, 2015-03-14:
This would enhance realtime troubleshooting on these channels.   I could have used this feature a number of times working on setups where we are using channels 9-12.

tridge, 2015-03-14:
When there are more than 8 input channels ArduPilot will send the RC_CHANNELS message with up to 16 channels. That is in all current releases. The problem is that MissionPlanner doesn't display it.
Please open an issue with MissionPlanner.

GrahamDyer, 2015-03-15:
https://github.com/diydrones/MissionPlanner/issues/812

rmackay9, 2015-03-16:
re-opening because although rc-inputs are covered, rc-outputs are not.

R-Lefebvre, 2015-12-22:
Attempt at fixing this is in here:

https://github.com/R-Lefebvre/ardupilot/commits/MAVlink_Servo_Output_WIP

However, this breaks MP, so new issue created:

https://github.com/diydrones/MissionPlanner/issues/1090

WickedShell, 2016-09-06:
This was addressed in 306487c (only I have misinterpreted either the issue or the commit).

closed 2016-09-06 by WickedShell