ArduPilot/ardupilot #1971: APM:Copter: fence action, what might happen

robustini, 2015-03-12:
With a "X8" (coax) we had an unpleasant event.
During an event "fence" at the speed of 14 m/s quad is literally bounced, resulting crash.
Analyzing the log we noticed excessive current consumption and an unexpected voltage drop, we are analyzing the hardware to understand if something is undersized.
Regardless this issues I believe that the "fence" is not operating properly, rather than engage immediately RTL, then force the quad has to fly in the opposite direction during the event , it would be correct that first should reach in Loiter for a few seconds (3 or 5, are enough) then later if the pilot does not engage in any action should automatically switch to RTL.
The video of this event:

https://www.dropbox.com/s/nsqi2hj6938103h/crash%2Bslomo.mov?dl=0

lthall, 2015-03-12:
Marco, you seem to be associating the fence action with the crash. This is misleading as any switch to Loiter, RTL, Auto, or even a sudden pitch back in Alt hold could have caused this crash. The name of this issue and associating the behavior of the fence with a crash seems a bit alarmist and misleading to me.

The fence isn't designed to tell you that you have reached a given range, it is intended to be outside anything you consider a reasonable/normal range and therefore only be broken by a fly away. It is therefore reasonable to switch directly into RTL as the pilot should not be deliberately flying into the fence. I would suggest that the simplest and most effective fix for this is to increase the fence to be larger than the range you wish to fly to.

We have been working on some collision avoidance code that will stop you flying into the fence in loiter. This may not be applied to Alt_Hold or Stabilize though.

R-Lefebvre, 2015-03-12:
I'm going to come in on the middle here.  I don't think fence should be an absolute limit, it gets advertised by some as a safe learning tool, so should not crash copters.  

However, it seems to me that this crash was just waiting to happen.  The copter tuning or power system obviously couldn't handle the sharp manoever.  This copter was going crash at some point, it just happened that the fence was the first sharp input it's ever seen.

rmackay9, 2015-03-13:
I've looked at the logs provided by Marco and done some tests in AC3.2.1 and master and it looks like we could do a better job of gently slowing the copter when entering RTL.
The first graph shows what happens in AC3.2.1 when entering RTL at high-speed.  Although better than AC3.1.5 it still slows down too rapidly and then pitches forward again.  In the actual log, the vehicle was travelling at a similar speed (14m/s in the actual logs, 18m/s in simulator)
[image]

The picture is worse in master in which we have a nasty spike when first entering RTL.
[image]

I'm going to title this issue.

rmackay9, 2015-03-13:
by the way, the vehicle's soft pitch tuning was certainly a major contributing factor to the crash.  A better tuned copter would have dealt with the sudden angle request without crashing.
[image]
[image]

lthall, 2015-03-13:
Remember the motors had stopped long before the copter hit the ground. So there are three issues here. The first is the power issue that caused the crash (That Marco already pointed out). And the second issue that I understand is the point of the issue is entering RTL when we hit the fence. 

The final issue and the one we should definitely work to improve is the overshoot when RTL is engaged. This may be as simple as a some tuning we should do better or we might be missing something in the way we are slowing the copter.

robustini, 2015-03-13:
Thanks team for your detailed feedback.
I have already alerted my friend to make a better tuning of its quad, considering also the excessive MTOW for that hardware setup.
However I believe is better slowing the vehicle more smoothly when entering RTL, because this thing could happen also during Auto mission at high speed for any reason (failsafe, manual switch, etc.).
If you remember more than a year ago during my tests  I reported the same problem, had been resolved but apparently has yet to fix better.

rmackay9, 2015-03-13:
I've fixed the new twitch that appeared in master so master should now act just like AC3.2.1.  So we can still do better but at least it's not worse!
[image]

robustini, 2015-03-13:
Thanks Randy for the super fast fix!

turfnet, 2015-03-13:
Great job

marcmerlin, 2015-03-19:
rmackay9 thanks for linking my bug here.
So just to be clear, would this fix the AP fighting my input so badly when it just switched to RTL and I'm giving input different from what it's trying to do?
https://www.youtube.com/watch?v=UBm7jdLaxiM&feature=youtu.be&t=5m42s (about 10 seconds) shows going from STAB to RTL for battery failsafe and how much the AP was fighting me for control until I just let go and let it fly back. The repeated shaking just feels wrong.

same thing here (RTL due to signal loss, watch about 20 sec)
https://www.youtube.com/watch?v=UBm7jdLaxiM&feature=youtu.be&t=0m55s

I think you said you fixed the part where switching to RTL causes this heavy 45 degree pitch on mode switch, thank you for that, but I still can't explain why it's shaking so badly when mixing my manual input with its RTL commands.
Will your fix also address this?

rmackay9, 2015-03-20:
@marcmerlin, once in RTL, the pilot has no control over the vehicle so that may explain why fighting with the vehicle for control didn't work.
http://copter.ardupilot.com/wiki/flying-arducopter/flight-modes/rtl-mode/

marcmerlin, 2015-03-20:
Hi Randy, I'm very perplexed, I've definitely had local control mixed with RTL.
1) I've more than once changed the spot my copter was landing at in RTL by moving it while it was landing
2) my videos show a lot of indecision and bouncing between 2 directions and those happened when my local control input was mixed with RTL. Once I let go, RTL worked, but when RTL takes over maybe because your TX signal dropped a second or two, that's very common to have both mixed and the results are quite bad as the video shows (I can definitely assure you that my TX input was not ignored when the X8 switched to RTL, it was mixed in).

Actually the 2nd video where I was flying back, and RTL took over and decided to rotate the X8 and find another way than me to go to the same RTL spot I was already flying back to, the fighting and stuttering that happened were quite bad.

Could you look into that if you can?

rmackay9, 2015-03-20:
@marcmerlin,
Ah yes, there are some exceptions sorry.  The repositioning during land is one of them, the other is that the pilot always has control of yaw in all flight modes.  So to be more accurate, the pilot has no roll and pitch input except for the landing stage of RTL.  So from the moment the vehicle switches into RTL, to the moment it begins descending over home, the pilot's roll and pitch input are ignored.
I've had another look at the RTL code and I'm pretty sure my understanding is correct. 
https://github.com/diydrones/ardupilot/blob/master/ArduCopter/control_rtl.pde#L358

marcmerlin, 2015-03-20:
Thanks for looking, that would explain why controlling in RTL mode during landing did easily do what I wanted while trying to control during other phases of flight seem to result in heavy shaking and the X8 fighting me until I let the sticks go.

So, I'm pretty damn sure that my sending input while the craft was in RTL was impacting it and causing the heavy shaking and misbehaviour I showed you in the videos.
http://marc.merlins.org/tmp/9410eaf3-9820-4911-a71e-4444fb71abf8.bin
has a full log so it should be easy to see when it first switched to RTL, my stick inputs, and erratic behaviour on angle and shaking.

Now, if I'm totally wrong, then we have another equally worrisome problem of figuring out why the X8 misbehaved for so long (20 seconds or more) after switching to RTL on more than one occasion.

Can you tell from the logs I gave?

Thanks, Marc

marcmerlin, 2015-03-24:
Actually let me follow up on my own post:
It looks like I was wrong in thinking that my stick input was being mixed with RTL input. I assumed that because of how unstable the copter was after it switched RTL and assumed it was fighting my input.

I met some 3DR folks today at ELC and found out the following which can explain the 2 time offsets I've seen and pointed out above:
#1: when switching to RTL, it seems that APM::Copter just ignores all flight info it could have gathered from the flying I just did in stab mode, namely where the compass points vs in which direction the copter is actually flying (that seems counter intuitive)

#2: when it switches to RTL, it basically starts from scratch and relearns to fly from the current state which could be off by a bit (seems that my compass was going off). This could be enough to explain why it just stops and flies sideways stuttering for a while before finally going in the right direction (except that I switched from RTL back to STAB)

At the 5mn42 offset, what threw me off is that it did the shaking thing again, and then turned 90 degrees, then shook again, and only then did it finally start flying to its RTL point. Apparently this is because the compass was off and it tried to turn the nose in the "right" direction, except that this was 90 degrees away from the real direction.

One suggestion I'd have if it's possible, is for APM::Copter to watch manual (STAB) flying and retune itself a bit for IMU and compass errors it can deduce from manual flying so that if it does have to take over, it doesn't start from a totally clean (and in my case, wrong), state.

Would that be possible?

marcmerlin, 2015-03-24:
Mmmh, then again I just checked my OSD data and the compass direction shown on the OSD is actually correct within about 10 degrees of the direction I was flying (unless the OSD shows GPS computed heading instead of compass heading in which case what I wrote above wrt to compass being potentially off by 90 degrees is still possible).

rmackay9, 2016-08-29:
Bumping to AC-3.5.

R-Lefebvre, 2016-08-29:
I think this might all be caused by having a flight speed higher than the set Waypoint speed or Loiter speed when entering these modes.  The initial "leash" calculation used to generated the stopping point,  uses the target speed, not the current speed.  So when the actual current speed is higher than new Flight Mode speed, the calculated stopping point is too close, and this results in the aggressive manoevering.

I see this if I have a high waypoint speed like 20 m/s, but Loiter speed at 10 m/s.  When I mission if completed, and it kicks you into Loiter, the copter stops way too quickly.  This has become a severe problem when flying my helicopter with the long-line payload.

At least, I think this is the cause, it's been a while since I looked.

Juzzle1, 2017-03-02:
Just wanted to chime in here with a very similar experience. 

Flying a fairly large X8 in stabilize mode, the pilot lost orientation of the copter and so decided to activate RTL. 

At the time RTL was engaged, the GPS / ground speed was 12m/s moving directly away from the home position.
While there was no crash in this case, the RTL immediately commanded maximum lean angle that led to an overshoot of almost 70 degrees(!)....... which was scary to watch! 

Thankfully, the RTL then worked flawlessly and brought the copter home.

In my case, I'm fairly confident this was caused by a soft tune combined with strong wind, (I had run autotune which came back with some very conservative rate roll / pitch P values (P=0.09)).

While I'll chalk this up to user error, I though the similarity to the OP's experience warranted a comment.

If anyone is interested in seeing my flight log let me know and I'll upload it.

Justin

[image]

Juzzle1, 2017-03-02:
I've just had another look at the log, and it would seem that the PIDs weren't actually that bad, especially considering how turbulent the air was that day.... although I'm sure someone more knowledgeable may say otherwise.

Logs attached hoping someone might have time for a quick look... @rmackay9?
[Juzzle1 RTL Overshoot.zip](https://github.com/ArduPilot/ardupilot/files/814969/Juzzle1.RTL.Overshoot.zip)

lthall, 2018-01-11:
This has been address in too many ways to list here. (ATT accel limits, mot throttle max being the main ones)

closed 2018-01-11 by lthall