ArduPilot/ardupilot #1972: Report un-truncated INAVerr as uint

jdennings, 2015-03-12:
There's been a number of conversations and confusion for users on forums, revolving around the fact that INAVerr, which counts GPS update misses,  is sometimes seen in post flight log analysis as non zero. (Copter; INAVerr does not seem at this point to be used on plane).  Log analysis shows that it is sometimes maxed out at 255 right from the start of a flight. in which case it is not possible to know wether  further in-flight misses happened.

This has mostly been observed by users flying m8n and derivative GPSs, known to be more sensitive to noise than earlier generation GPS. In most cases, the errors accumulate at start-up, which seems to indicate they can be ignored. More troubling  in-flight INAVerr increments, however, have also been seen, both with m8n and  with 3DR and other 6 generation GPS, with no apparent detriment to flight quality. 

This enhancement would allow better post flight analysis, help determine/quantify  wether high INAVerr counts are an issue, and further shed some light on the adequacy of higher end GPSs, which so far have been thoroughly tested by some, have been observed to offer better precision, been flown very successfully, yet  appear to generate more errors and according to some analysis are not recommended for use.

http://www.rcgroups.com/forums/showpost.php?p=31027087&postcount=3716
http://www.rcgroups.com/forums/showpost.php?p=31030526&postcount=4041
http://diydrones.com/forum/topics/arducopter-3-2-1-beta-testing?id=705844%3ATopic%3A1882038&page=30#comments

jdennings, 2015-03-12:
Should have added: Not recommended for use "without further shielding".

koumac, 2015-03-12:
This error is reported under PM, Performance Monitor
What'  the meaning of  "INAVERR"  error ? 
Is the the INAVERR 32bits , which leads to a max value of 255 ?

rmackay9, 2015-03-16:
This is a bit unlikely to get done (at least as stated) because we're moving to the EKF for AC3.3 and errors are reported very differently.  Instead of a simple INAV error counter there's extensive (but perhaps not yet well understood by the community) reporting on "variances" which are a 0.0 ~ 1.0 representation of how good the EKF thinks it's various attitude and position estimates are.

jdennings, 2015-03-26:
I was going to say "but what about DCM" but read your post about smallEKF and wha sounds like the  upcoming disappearance of DCM. But maybe a new counter in this case, something like INAVupdates, separate from INAVerr that simply counts missed updates in the GPS reading loop? This would be useful imho.

rmackay9, 2015-08-21:
Copter-3.3 is about to be released and as expected Inertial Nav has been replaced.  The EKF comes with much more detailed reporting of "innovations" which reflect the health of the system.  Still, I've created this other issue to capture the logging of GPS and Baro glitches which I think is less obvious now than it was in AC3.2.1 (we can still see the glitches by looking at the level of the innovations but they don't stand out in the logs like ERR messages do).
https://github.com/diydrones/ardupilot/issues/2729

closed 2015-08-21 by rmackay9