ArduPilot/ardupilot #1974: Flight Mode Channel Should Be Selected By Pilot

publiux, 2015-03-12:
In ArduCopter, the flight mode is selected via channel 5 by default.  However, this channel should be selectable by the user but default to channel 5.  When using a PPM receiver, channel 5 can sometimes not be bound to to control settings that would yield 6 flight modes. For example, the Turnigy 9X can create 6 flight modes by mixing channels, but the mix will be output over channel 6. Using a PPM receiver, channel 6 will not change the flight mode.

magicrub, 2015-03-13:
Plane has "FLTMODE_CH" which allows you to select the PWM/PPM channel
input, I don't see why copter can't do the same thing.

-TomP

ausi1972, 2015-06-11:
It would be useful to select the flight mode channel and also swap ch6, 7 & 8 as well so you can use ch6 for RTL or auto tune etc. So people using non programmable radios like the DJI DT7 Tx with DR16 Rx (which is a lovely radio for flying multi rotors) can use the flight mode switch, camera tilt roller and RTL switches for the labeled features on Arudcopter.

Rk-Hk, 2016-06-24:
ausi1972 Se puede volar el apm2.6 con DR16 rx , si la respuesta es si me podrías ayudar , es para un proyecto de curso , te agradecería muchísimo !! gracias de antemano. #

magicrub, 2016-06-24:
@Rk-Hk please use discuss.ardupilot.org or the gitter chat channel for support questions/requests

DavidIngraham, 2018-05-04:
Resolved, Copter 3.6 now has  FLTMODE_CH

closed 2018-05-04 by DavidIngraham

CrazyProfessor, 2019-06-24:
Hello there, 
Is there any way to add this FLTMODE_CH to V3.2.1? My FC is unable to upload V3.6. I am having the exact same problem as described above RE: I would  like to have flight modes on Ch6 not 5. Thanks

rmackay9, 2019-06-25:
@CrazyProfessor, I think someone replied in another thread on this same topic but I'm afraid we can't backport this feature to 3.2.  It's all open source of course so the code is there and anyone is free to try but from a core dev team perspective it's too large a jump back and we're not supporting new releases of Copter-3.2.x.

A much better solution is to buy a more [modern flight controller](http://ardupilot.org/copter/docs/common-autopilots.html).

CrazyProfessor, 2019-06-26:
Thank you for your help. Is it possible to flash a different brand of firmware onto the APM so I can get the mapable flight modes? eg: Px4, or inav?

rmackay9, 2019-06-26:
@CrazyProfessor, I really don't know about other systems sorry.