ArduPilot/ardupilot #1975: Copter: SIGFPE crash while taking off in POSHOLD (SITL)

claymation, 2015-03-12:
As of 007c96a3d8afa82fb58af92c7e54d9e95294bf4b, attempting to take off in POSHOLD mode crashes with:

    Program received signal SIGFPE, Arithmetic exception.
    0x00000000004586a0 in AP_MotorsMatrix::output_armed (this=0x702ec0 <motors>)
        at /home/clay/ardupilot/libraries/AP_Motors/AP_MotorsMatrix.cpp:293
    293                 rpy_scale = (float)(out_min_pwm-thr_adj-out_best_thr_pwm)/rpy_low;

The culprit: `rpy_low` is zero.

Steps to reproduce:

  * Check out 007c96a3d8afa82fb58af92c7e54d9e95294bf4b
  * `sim_vehicle.sh -G --console --map --aircraft test`
  * `arm throttle`
  * `mode poshold`
  * `rc 3 1800`

claymation, 2015-03-12:
To be honest, I'm not sure if that's a valid use case, but I use it regularly in SITL to keep the copter centered over a simulated landing target while developing some vision-aided navigation software (yet another precision lander).

rmackay9, 2015-03-13:
@claymation, thanks for that I'll try and reproduce.

rmackay9, 2015-03-13:
Ok, I've reproduced it!  We will fix asap!

rmackay9, 2015-03-16:
This should be fixed in master now, thanks!
https://github.com/diydrones/ardupilot/commit/31a55b2bd61a0aa147b4033800c064f03125f656

closed 2015-03-16 by rmackay9