ArduPilot/ardupilot #1976: throttle can unsuppress when we get GPS lock

tridge, 2015-03-12:
if in RTL on ground and get late GPS lock throttle in plane can unsuppress

magicrub, 2015-03-13:
I have experienced a throttle unsupressing itself while handling the aircraft during a bungee TAKEOFF which is quite dangerous but was never able to reliably duplicate it. Even happened once while sitting on my bench - would be in line with a GPS loss. I'll try testing /w intentional GPS loss and report back.

magicrub, 2015-03-13:
@tridge you had mentioned on mumble how you may be able to duplicate bug's behavior in SITL but my memory is failing me, what was that procedure?

magicrub, 2015-03-14:
while looking over the code I notice that if you're armed and in a TAKEOFF waypoint, such as a bungee launch, the throttle suppress can come off due to an altitude or gps glitch. This is specific to the TAKEOFF wp, and a little different than what you're observing, but it would probably be a good idea to change suppress_throttle()
from:
    if (control_mode==AUTO && 
        auto_state.takeoff_complete == false &&
        auto_takeoff_check()) {
            // we're in auto takeoff
            throttle_suppressed = false;
            return false;
    }

to:
    if ((control_mode==AUTO) && (auto_state.takeoff_complete == false)) {
        if (auto_takeoff_check()) {
            // we're in auto takeoff
            throttle_suppressed = false;
        }
        return false;
    }

That way if we're ever in a takeoff situation where people are handling the aircraft while armed we should ignore any GPS/altitude glitches regardless of how they happened. When in takeoff, the only way to unsuppress is to satisfy auto_takeoff_check() which has timers and such.

magicrub, 2015-04-27:
This may be fixed, please confirm @tridge

magicrub, 2015-06-10:
confirmed, this is now in master.

closed 2015-06-10 by magicrub