ArduPilot/ardupilot #1977: Copter: Oscillation in POSHOLD mode in recent master

claymation, 2015-03-13:
Somewhere between 9871b955866ed21e9c13b7f857222d17fa33aebd and 88ec13b10d913d72cdb0b24ba2e1244e6ed37734, a change was introduced that causes a ~4 Hz / ~3 degree oscillation in pitch and roll for copters flying in POSHOLD mode (at least in SITL). The vehicle maintains position, but wobbles, as if balancing on the tip of a pin.

[image]

/cc: @jschall

rmackay9, 2015-03-13:
Thanks for the report!
I think I've found and fixed the cause here.  I'm going to close this issue but will re-open if you say it's not fixed.
https://github.com/diydrones/ardupilot/commit/bb74b8dec86b074522438f2b83aff7ec5070d3be

[image]

closed 2015-03-13 by rmackay9

claymation, 2015-03-13:
@rmackay9 thanks for looking into it, but if anything, it's worse now:

[image]

I can make a screencast of how I'm running the sim if that would help, but basically it's just:

  * `git checkout bb74b8d`
  * `make clean`
  * `sim_vehicle.sh -w`
  * `level`
  * `sim_vehicle.sh -G -L ATL --console --map --aircraft test` (where I've defined ATL to be near my house)
  * `arm throttle`
  * `rc 3 1500`
  * After ~1–2 secs:
  * `mode poshold`
  * `rc 3 1800`
  * Once we get to ~25m:
  * `rc 3 1500`

I can see the pitch and roll oscillating in the mavproxy console, which shows up in the graphs post-flight.

How can I help track it down?

claymation, 2015-03-13:
Here's another log from a flight where the oscillation didn't start right away:

[image]

Is there a handy place I can dump the .BIN file for you?

claymation, 2015-03-14:
@rmackay9 I just ran into this on an older revision (), too, so I'm wondering if there's just something weird about my SITL setup. I noticed this in the docs:

> SITL can also be run within a virtual machine on Windows, Mac OSX or Linux, although this can cause timing issues and make it difficult to fly.

I'm running SITL on Ubuntu within a VMware Fusion VM with four cores allocated to it—do you suppose SITL just doesn't like running in my VM?