ArduPilot/ardupilot #1980: Copter:Pixhawk:(Hexa/Y6 Specific) When 6 motors are enabled, servos enabled on Chan 7/8 Out are damaged

teamtiltrotor, 2015-03-13:
Problem: While developing motors_x.cpp files for tiltrotor, it was discovered that when 6 motors are enabled on output channels 1-6 out (Pixhawk), that output channels 7 and 8 results in servo buzzing, overheating, and servo failure in minutes. 
  
Troubleshooting: 
1. This problem was duplicated on Copter:Y6 and Copter:Hexa SW version 3.2.1 with output channels 7 / 8 set to camera gimbal servos. 
2. Changing the ESC update speed (RC_SPEED) and Servo Update Speed MOT_SV_SPEED does not change the behavior.
3. This was duplicated on 3 different Pixhawk units with no common hardware
4. Can be duplicated with all ESC's removed from the output rail, and only power and servos connected
5. Problem does not exist on chan 7/8 when 4 motors are enabled (QUAD). 
6. Servo buzzing does not exist on heli, single, or tri-frame (any copter frames with servos), and changing MOT_SV_SPEED is effective.

Cause:Unknown

Impact: Minimal - using aux_servo outputs 9-14 is our current work around.

rmackay9, 2015-03-14:
This will definitely be related to the RC output speed on those pins.  So it'll be at 400hz for the motors but for servos it needs to be down around 125hz.

rmackay9, 2015-08-22:
I don't think we can fix this in software on the Pixhawk, it's really a hardware issue so I'm going to close it.

closed 2015-08-22 by rmackay9