ArduPilot/ardupilot #1986: APM for Linux does not compile natively

vmayoral, 2015-03-17:
Hi @tridge, just found out that it's not possible to compile the code in the BBB anymore:
```
...
%% libraries/AP_HAL_VRBRAIN/UARTDriver.o
%% libraries/AP_HAL_VRBRAIN/Util.o
%% libraries/AP_InertialNav/AP_InertialNav_NavEKF.o
%% libraries/AP_InertialSensor/AP_InertialSensor_Backend.o
%% libraries/AP_InertialSensor/AP_InertialSensor.o
/root/ardupilot/libraries/AP_InertialSensor/AP_InertialSensor.cpp: In member function ‘bool AP_InertialSensor::calibrate_accel(AP_InertialSensor_UserInteract*, float&, float&)’:
/root/ardupilot/libraries/AP_InertialSensor/AP_InertialSensor.cpp:521:29: internal compiler error: Segmentation fault
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.6/README.Bugs> for instructions.
```

I tried switching into version 4.7 and i got the same result. There're no debs available for armhf  g++-4.8 but will try compiling it from source.

vmayoral, 2015-03-17:
Tried it with `g++-4.8` (and 4.9). It solves this issue but it seems relevant the fact that older compilers do crash.

srik901, 2015-03-18:
Can confirm that this breaks for make sitl
It works fine for make px4-v2

aaraujo11, 2015-03-21:
Hi! I also confirm it, it's not possible to compile the code in the BBB anymore. 
```
/home/debian/ardupilot/libraries/AP_InertialSensor/AP_InertialSensor.cpp: In member function 'bool AP_InertialSensor::calibrate_accel(AP_InertialSensor_UserInteract*, float&, float&)':
/home/debian/ardupilot/libraries/AP_InertialSensor/AP_InertialSensor.cpp:521:29: internal compiler error: Segmentation fault
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.6/README.Bugs> for instructions.
Preprocessed source stored into /tmp/ccdBNiaU.out file, please attach this to your bugreport.
make: *** [/tmp/ArduCopter.build/libraries/AP_InertialSensor/AP_InertialSensor.o] Error 1
```

@vmayoral how do you compile with g++ 4.8 or 4.9? I already install the g++ 4.9 changing the /etc/apt/sources.list to jessie version to be possible install it (once it doesn't exist in wheezy repository), even update the namespace to the makefile use the g++ 4.9, but without success.  I'm using Debian wheezy 7.5 with kernel 3.8.13-bone50.

vmayoral, 2015-03-21:
@aaraujo11 modify https://github.com/erlerobot/ardupilot/blob/master/mk/find_tools.mk

aaraujo11, 2015-03-22:
Thanks @vmayoral , i had just modify the NATIVE ones not in the toolchains for beagleboneblack, it compile perfectly.

vmayoral, 2015-03-22:
@aaraujo11 np, which sensor cape are you using? Did you try the [PixHawk Fire Cape](https://erlerobotics.com/blog/product/pixhawk-fire-cape/) or [Erle-Brain]((erlerobotics.com/blog/product/erle-brain/))?

magicrub, 2015-04-27:
is this still an issue?

AkshayBudhiraja, 2015-05-18:
Yes @magicrub, tried setting SITL up today and I'm facing the same problem. Had to make it use g++-4.9

duststorm, 2015-10-22:
I just bumped into this issue as well, while trying to compile SITL on raspberry pi. So it appears that with the latest version this is still a problem.
Luckily there is a fairly easy way to get g++4.9 on raspbian: https://solarianprogrammer.com/2015/01/13/raspberry-pi-raspbian-install-gcc-compile-cpp-14-programs/

To make it compile with g++4.9 I add the following line to mk/board_native.mk:
```
# On raspberry make sure we use a recent compiler that supports c++11
CXX = /usr/lib/ccache/g++-4.9
```
(yes there might be more elegant solutions to this but this was a quick fix that worked for me)

For some reason the logic in sim_vehicle.sh that uses dirname for finding the autotest folder does not work, to get around this I run the sim_vehicle.sh script from the ardupilot/ArduCopter folder.

lucasdemarchi, 2016-02-11:
Minimum required version is now 4.8 (4.9 recommended).  By setting the CXX variable is even possible to change the compiler detection with the waf build system (for example to use another compiler version or clang++). Compiling for sitl should work just fine as long as you have the required version. So I'm closing this old bug. Feel free to open another one if it's not working for you.

closed 2016-02-11 by lucasdemarchi