ArduPilot/ardupilot #1987: AUX pass through doesn't work when using SBUS with the failsafe bit set.

cmanley, 2015-03-17:
I'm using Arducopter 3.2.1 (but this may apply to plane & rover too), a Pixhawk, and FrSky L9R receiver with all 12 channels over SBUS.

Throttle failsafe is set up correctly so that if the receiver browns out, then that's covered. That works.

But because Mission Planner & Arducopter don't have user definable failsafe events list feature (yet), I configured my FrSky L9R to use channel presets failsafe so that when a failsafe occurs, these additional things are supposed to happen:
CH5 mode changes to RTL
Lights and strobe go on (CH10 high)
LMA (lost model alarm) goes on (CH7 high)
Camera switch selects FPV camera (CH9 low)

The LMA is connected directly to the receiver's CH7 port (CH7 has no assigned function in my Mission Planner).
The lights are connected to the Pixhawk's AUX2 with RC mode set to pass through. This is tested and it works.

When I switch my radio off, then this happens:
1. The LMA (directly connected to receiver) beeps indicating that receiver failsafe presets are activated.
2. The lights do not turn on, indicating that Arducopter isn't reacting as expected to the failsafe, even though I verified by using an SBUS decoder that the receiver is really setting CH10 high.

I analysed the log file containing the failsafe event, and what I see is that CH10 does go high on the RCIN, but the Pixhawk (or Arducopter) simply doesn't pass the new value through the AUX2 port anymore. 

Here is a screenshot of the relevant log data:
http://ardupilot.com/forum/viewtopic.php?f=100&t=11652&p=30656#p30684

B.t.w. displaying RC inputs & outputs > 8 in MP will make debugging a lot easier: https://github.com/diydrones/MissionPlanner/issues/812

P.S. I downloaded the source code and tried to find where this is going wrong, but without guidance, I think it's a bit too challenging for a mere mortal such as myself to build a mental picture of how all the code fits together and where to focus.

apphead, 2015-09-12:
I can confirm the described behaviour in a similar setup with Arducopter 3.2.1 as well.
Landing gear connected to 
- L9R receiver -> failsafe (by receiver presets) works
- Pixhawk AUX1 with passthrough works in non-failsafe mode (so config must be ok) but fails to react when passed thru RC9 in Pixhawk.

(PS: Why not include the landing gear function from CH8 Option in all the AUX Channels as well? The way it is solved in 3.3RC? two channels must be assigned: CH8 for the input and CH9 .... CH14 for PWM output. So, one channel goes to waste which is a shame)

aldoir, 2016-05-12:
I can confirm this behaviour on Copter 3.3 as well.

All gimbal channels are ignored when failsafe is on

Naterater, 2019-01-07:
For all of this, what is THR_FS set to? 0 or 1?

rmackay9, 2019-01-08:
I think we should close this actually.  Copter has a user configurable list of responses now so the user shouldn't be setting up the radio failsafe as described.  The versions of Copter tested are quite old and I think it is the expected/desired behaviour that during a radio failsafe none of the pilot's transmitter inputs are used including auxiliary switches.

closed 2019-01-08 by rmackay9