ArduPilot/ardupilot #1989: Sudden height change in ALT HOLD - dangerous

vmayoral, 2015-03-18:
@rmackay9 and @tridge,

We'd appreciate if you could take a look at [this](https://github.com/erlerobot/logs/blob/master/18-03-2014/althold_error.BIN) log.

@imuguruza and @ahcorde noticed today (it seems it happened several times) that when taking off in `ALT HOLD` the copter suddenly gets height really aggressively:

[image]

Changing to STABILIZE allowed us to land the copter Has someone else experienced this?

R-Lefebvre, 2015-03-18:
Hi Victor, I'm having a look at it.  So you're using 3.3-dev, appears to be recent, but not the latest.  I believe it's using the new IMU filtering scheme as you have INS_GYRO_FILTER and INS_ACCEL_FILTER parameters.  But you also still have AHRS_EKF_USE which I think has been deprecated?  BTW, this was not 1, so I believe you were using DCM rather than EKF.  You should probably be using EKF.

I do see a permanent, and remarkably consistent offset between the Baro height, and the inertial height estimate.  It actually begins before the motors start.  Normally this sort of problem is caused by vibrations, but that doesn't seem to be the case here.  The vibration levels look good, and as I said, the divergence between INAV altitude and baro altitude begins before the motors even start.

I do notice that your IMU is registering only 8 m/s/s acceleration downward while sitting still.  The system basically thinks it is accelerating downward. This is probably the reason for the offset.

I think something went wrong during your accelerometer calibration.

vmayoral, 2015-03-18:
@R-Lefebvre thanks a lot for taking a looking that up. Just saw what you mean. I'll leave a capture here in case someone experiences something similar.

[image]

vmayoral, 2015-04-15:
Some conclusions posted [here](https://groups.google.com/forum/#!topic/drones-discuss/IIbxvqpgrP4).

in a nutshell, it seems to us that these misbehaviors are due to a bad set of parameters (we are using [these ones](https://github.com/erlerobot/erle_parameters/blob/master/copter/copter_autotune_th9x.param)).

For completion, falling back to DCM ([forcing it in the code](https://github.com/erlerobot/ardupilot/commit/a1ec4f850ea92bbd23175b59eb00aea7356d7de0)) fixes this sudden changes.

rmackay9, 2015-04-15:
@Victor,
Looks like you've got IMU (accel, gyro) and compass offsets in that param list.  That's fine as long as you're always using the same board.

victor, 2015-04-15:
those voices in my head... please make them stop!

vmayoral, 2015-04-16:
Thanks @rmackay9 for your answer.

We also noticed that while armed, we can't change to GPS-enabled modes (e.g.: we can't change from `ALT HOLD` to `LOITER`). Any chance this is being cause by a missconfiguration in the EKF?

rmackay9, 2015-04-16:
@vmayoral, it's all guesswork without a log file.

vmayoral, 2015-04-19:
@rmackay9 rebased the code including latest commits from @priseborough and redid the tests. Looks much better. We can now change modes (we were previous getting `EKF VARIANCE` every single time) [Here's a log](https://github.com/erlerobot/logs/blob/5bde6a8b6fd5fba4dffb072ead3e15985eb7022b/pxf/copter/19-4-2015/8.BIN).

We'll keep looking into it to make sure issues have been fixed.

rmackay9, 2015-04-20:
ok great.  let's close this issue then and we can re-open if it reappears.

closed 2015-04-20 by rmackay9