ArduPilot/ardupilot #1993: Plane: Auto takeoff exceeds parameters and does not transition to next waypoint smoothly

WickedShell, 2015-03-19:
The plane exceeds the pitch parameter entered in the mission planning (15) and holds 20 or more pitch up the whole time, and exceeds the set cruise airspeed (18m/s) and exceeds 20+m/s during the takeoff cliimb. This mostly seems to be related to holding in TKOFF_THR_MAX at all times, even after exceeding the desired takeoff parameters. TECS_PITCH_MAX is 0 which I'm given to understand means it reads the takeoff angle from the waypoint, but it doesn't appear to.

The other problem is that after hitting the target altitude the plane always pitches over in a dive, and kills throttle. It will usually pull out of the dive at around the target altitude for the takeoff but will then need to climb to continue to the next waypoint. For example the takeoff waypoint has an altitude set of 60 meters and the next waypoint has 90 meters, this results in the plane entering a diving turn towards the new target waypoint.

A relevant log file can be found here, including 3 auto takeoffs: http://files.swiftradioplanes.com/116.BIN (36MB)

magicrub, 2015-03-19:
I too experience the dive after takeoff had completed on every launch. The
other pitch stuff works fine for me though.

tridge, 2015-03-19:
The takeoff pitch is specified in the mission, not in parameters. 
The throttle is indeed always TKOFF_THR_MAX. You need to adjust to suit.
The dive on ending takeoff is indeed an issue. What we need is to smoothly cut over between takeoff and normal flight, in a similar way to what has been recently added for the flare transition in landing.

WickedShell, 2015-03-19:
So the plane exceeding the desired pitch on climb is simply due to to much power being applied?

Will there ever be adaptive throttle for takeoff? We are using a 100% throttle as we handlaunch the plane and need all the power for the first few moments of flight. On a manual takeoff 100% throttle is applied until the aircraft starts to climb, at which point it can easily be reduced to 80%. Using less then 100% throttle would also be good, as the batteries we fly with are normally experiencing minimal draw, but on takeoff we are inside out 15 second burst power limit.

If the plane doesn't dive then the lack of throttle control would probably be fine, as I could lower the takeoff altitude to a lower value and allow normal climb logic to takeover at that point if I can safely expect the plane to keep climbing.

tridge, 2015-03-19:
the pitch being off from the value in the mission will be because the pitch tuning is quite good enough. Probably too small I term.
Yes, we should support variable throttle in takeoff. We should slowly reduce throttle once cruise airspeed is reached.

WickedShell, 2015-04-15:
This is primarily fixed by 78a3ce4 and 4b1d71e helps as well. At this point the aircraft I've test with now climb out at the maximum tecs land rate, and no longer results in overspeeds or dives.

closed 2015-04-15 by WickedShell