ArduPilot/ardupilot #1995: Scripting

abvgeej, 2015-03-19:
At least some scripting support (if X and Y, then Z) is a feature which we all need, and in my opinion needs no comment.

magicrub, 2015-04-05:
This is extremely useful but also extremely difficult to get right. It's even harder to do it in a way that will satisfy a general audience rather than a single-purpose hack. I'm open to suggestions as to how to do it or at least how the user interface could look like.

rmackay9, 2015-04-14:
I think it's unlikely we will add scripting within the ardupilot code.  Instead we're moving towards DroneKit which allows a companion computer or ground station to run scripts.  http://dronekit.io/

I think I'll close this because although the general request is valid, it's not something we would add to ardupilot (which is what this issues list is for).

closed 2015-04-14 by rmackay9

abvgeej, 2015-04-14:
No good autopilots todays. Eh....
And yes, its a very difficult feature to get right, and all i can do now is to sit, and think more about getting a ErleBrain.
(Yet, i would say that the code is nice and clean enough to get the features you need when you do, but in-air)

rmackay9, 2015-04-15:
A Linux board (like the ErleBrain) or a companion computer is a good idea... probably the best idea to get scripting working in the air.  This is the model that the 3DR solo is following and what I've personally used for things like the red-balloon-popper.  I think it's the way to go.