ArduPilot/ardupilot #1998: Dataflash logging drops messages without any explicit indication in high-load situations

hsteinhaus, 2015-03-20:
Some flights with LOG_BITMASK set to -1 and a camera triggered by ArduCopter revealed that log messages can get lost (camera had pictures that did not correspond to an appropriate CAM message). During this flight, about 1% of all CAM messages got lost.

During tests with reduced logging but identical hardware and software no CAM messages got lost. The SD card (original 3DR) did not show any performance or block error issues.

Currently it is not possible to detect this situation by looking on the log only. This effect can make log analysis hard to impossible in certain situations. Additionally, the current behavior can seriously mess up any kind of mapping mission, that relies on a complete camera trigger log for geo-referencing of the taken images.

See also here: https://groups.google.com/forum/#!searchin/drones-discuss/hickup/drones-discuss/tlE_PRpuW4w/b1fD2MCgdswJ

wjax, 2015-03-20:
Same happens in my pixhawk setup with AC 3.2.2.

Normal mission takes about 700 pictures. CAM msg loss is either none, 1 or 2 maximum.

hsteinhaus, 2015-03-20:
Forgot to mention that I used ArduCopter 3.2.1 . 

The spatial jitter of the CAM messages is excellent, (<<1m difference from configured interval), so I guess this issue is not related to cam triggering.

R-Lefebvre, 2015-05-15:
Wonder if this can be closed?  We are no longer developing APM2?

rmackay9, 2015-05-17:
Yes, I think we should close this.
It's not a surprise that full logging (i.e. LOG_BITMASK = -1) on the overloaded APM2 leads to a few lost messages.  I think the work-around is to reduce the logging.  For example turn off the IMU logging and fast ATT logging.

closed 2015-05-17 by rmackay9