ArduPilot/ardupilot #1999: Waypoints on SD card

VladimirP1, 2015-03-20:
Why does Pixhawk need so many flash memories?
I mean you could store waypoints on SD card also. (Can even leave the same memory layout, only increase size and put to binary file in SD card)

It will be possible to fly in loiter, then "replay" the flight (For example load waypoints to it, generated from tlogs)

rmackay9, 2015-03-20:
SD cards are not that reliable actually because the board can be powered down at any time which can corrupt the FAT file system.

Copter has the ability to store the current location as a waypoint using a ch7/ch8 switch although the recording of the points is done manually (it doesn't automatically record them as the vehicle flies).

May this is more like a question that could go on drones-discuss@googlegroups.com?

VladimirP1, 2015-03-20:
Does NuttX have partitioning support?

VladimirP1, 2015-03-20:
Terrain following also writes to SD card.

magicrub, 2015-04-27:
What is the status of this issue?

rmackay9, 2015-04-28:
I think we can close this.  We support up to 700+ waypoints on Pixhawk now so not sure how much we would really gain by moving to a less reliable storage

closed 2015-04-28 by rmackay9