ArduPilot/ardupilot #2005: Dead Reckoning if GPS Lost

MetalSlugX, 2015-03-23:
For now, we only have the possibility to either use dead reckoning or gps, how could one use GPS for the Mission then if the GPS Signal is Lost (Clouds, Plane going indoor, Interferences) change into Dead Reckoning Navigation?

    // @Param: GPS_USE
    // @DisplayName: AHRS use GPS for navigation
    // @Description: This controls whether to use dead-reckoning or GPS based navigation. If set to 0 then the GPS won't be used for navigation, and only dead reckoning will be used. A value of zero should never be used for normal flight.
    // @Values: 0:Disabled,1:Enabled
    // @User: Advanced
    AP_GROUPINFO("GPS_USE",  3, AP_AHRS, _gps_use, 1),

lthall, 2015-03-23:
This has been covered many times before. We can use dead reckoning for about 5 seconds before the errors get too large. Our sensors simply aren't accurate enough to do any better.

So this one can be closed.

MetalSlugX, 2015-03-23:
I would disagree, the new EKF code with PxFlow Sensor could estimate the actual Position with much higher accuracy than just using accelerometers & gyros.
The question is, where in the code could one change the settings for at least, if GPS is lost, use dead reckoning for RTL.

rmackay9, 2015-03-23:
@MetalSlugX,
AC3.3 (and master) support the px4flow sensor but besides enabling the px4flow sensor no additional changes are required.
http://copter.ardupilot.com/wiki/common-optional-hardware/common-optical-flow-sensors-landingpage/common-px4flow-overview/

closed 2015-03-23 by rmackay9