ArduPilot/ardupilot #2008: Codes doesnt compile for APM1 on linux

zjcui, 2015-03-23:
Hello team,

I try to compile the code for my apm1 chip (mega1280) on my ubuntu laptop. I modified the config.mk as:
BOARD = mega
AL_BOARD ?= HAL_BOARD_APM1

but can not make it, one of the errors is like:
error: 'class AP_InertialSensor' has no member named 'calibrate_accel'
                 if(ins.calibrate_accel(&interact, trim_roll, trim_pitch)) {

I tried ArduPlane, ArduCopter and APMRover2, but no luck.

Please let me know what to do. Does the code here still support APM1 (mega1280 board), if so, how to configure(i guess i need to disable something which i dont know). 

THANKS
Zhengjie

taedla01, 2015-03-23:
Hello @zjcui,

Actual code from master do not fit on old APM1 boards with AVR 1280 chips. Only AntennaTracker still supports that board and it is mandatory to do some changes on code as it does not fit natively. ArduRover fits but with several deep changes on code.

Cheers,
Dario.

zjcui, 2015-03-23:
Hello Dario

Thanks for your reply. Just wondering is there any code i can use for this APM1? Or some previous version still available on the website for this chip? 

Thanks
Zhengjie

taedla01, 2015-03-23:
Hi @zjcui !
On http://firmware.diydrones.com you can search for latest .hex for all boards. For example, for Arduplane (latest for APM1 1280):
http://firmware.diydrones.com/Plane/latest/apm1-1280/

And if what are you looking for is the "source code", I remember that last native version for 1280 was "Arduplane 2.34" but I do not know where do you can find it...sorry. Other recent versions like 2.60 fits on 1280 but you need to change code.

Regards from Spain,
Dario.

zjcui, 2015-03-23:
Hi Dario

yup, I am looking for the source code and code for our own cases
Alright, anyway thank you so much

Zhengjie

rmackay9, 2015-03-24:
@zjcui, The ArduCopter-3.2.1 tag should get you the last version that the APM1 support.
https://github.com/diydrones/ardupilot/commit/36b405fb0b2b3ef77952e8c9f170dbcf1976cb30

closed 2015-03-24 by rmackay9