ArduPilot/ardupilot #2010: Arduplane check_require INS issue

magicrub, 2015-03-24:
ArduPlane+Pixhawk+EKF

See attached log.

I'm having trouble with ARMING_REQUIRE=16 (INS), the EKF4:FS failure flags=128 which means EKF is not initialized. I have a 3D lock and everything is calibrated. Works fine in ArduPlane v3.2.0 but master won't initialize.  I've seen other related problems such as gyro errors and inconsistent accels too.

Seen this on several pixhawks so not a hardware problem.

magicrub, 2015-03-25:
This behavior was seen on @tridge 's ELC demo where he was unable to get a healthy AHRS and had to roll back to an earlier version to successfully fly. I've narrowed it down to the flurry of commits between Dec 17, 2014 and Feb 5, 2015.

mday299, 2015-03-26:
Here, here.  We're seeing similar problems -- most evident on the bench, not in SITL.

magicrub, 2015-03-27:
for us, workarounds are:
turn off EKF
power-down pulsed-lite LIDAR

tridge, 2015-03-28:
I've pushed a fix for this to master. Thanks for the log!

closed 2015-03-28 by tridge