ArduPilot/ardupilot #2014: Support for stall avoidance in APM::Plane while in MANUAL mode

marcmerlin, 2015-03-25:
I had a chat with Tridge, and I understand that if you fly in FBWA, the new APM::Plane code can do stall avoidance for you.
My suggestion was to allow this code to run in a mode based off MANUAL, maybe a new mode since STABILIZE does get in the way of manual flying a bit.

Basically the new NOSTALL mode would work just like MANUAL until the flight computer thinks that the airframe is about to stall.

4 scenario I can think of
A) flying level
If the motor is off and altitude allows, it would just lower the nose and allow the rest of the input. Ideally it would output something on the OSD to say it did that.
If motor is on, add throttle and/or lower the nose.
If the motor is off and altitude is low, turn the motor on if any, or lower the nose.
(yes, you have to worry about an intentional stall to land, if there is no good source of altitude to know you're less than 1m over the ground, maybe the pilot would have to switch back to manual to be able to stall for landing)

B) commanding a turn
a) if high enough, allow the turn but lower the nose and/or power up (if motor is already running)
b) if not high enough, decrease the bank angle to avoid a tip stall and spin to the ground.

Why would this be useful?
I fly a multiplex cularis, I love flying it the normal way (MANUAL) but it will tip stall easily depending on how loaded it is with weight and when I do a manual landing with base to final turn, it's just hard to know my airspeed and how close I am to stalling and crashing.
Example: https://youtu.be/K9hkkJ6Ypbw?t=7m49s

Thanks :)

magicrub, 2015-03-25:
I too have that problem when flying in manual and stabilized. However,
manual needs to stay as-is for the sake of it being called manual. However,
instead of a new mode, I wouldn't mind seeing a throttle based steak
prevention added to stabilized.

Meanwhile, fbw should suffice.

marcmerlin, 2015-03-25:
First, I agree that MANUAL should stay manual, enhance my suggesting a new mode called NOSTALL.
Tridge kind of talked me out of using STAB, and re-reading its description I'm not sure I'd like it :)

In the meantime, you are correct that using FBWA will work. I'm a full size pilot however and I'm a bit wary of teaching my brain to fly the wrong way with regards to ailerons, but I may have to use that anyway for now if I want stall protection.

Cheers.

magicrub, 2015-04-05:
perhaps we can have a param like "STAB_PREVENT_STALL" that would do some (minor?) work to prevent stalls such the ones you suggest. It would not be comprehensive like FBW is but could help a little.

marcmerlin, 2015-04-06:
@magicrub Adding this to STAB would be great (actually you can argue it would very much make sense), but I would love a mode that handles like MANUAL, while simply adding stall prevention.
Hence, maybe a new mode like NOSTALL (previous comment).
Why?
1) just like stick shakers in real jets. They fly like normal planes, but the plane will put the stick down for you if you're about to stall
2) RC pilots used to normal RC flying (i.e. MANUAL) are likely to keep wanting to fly that way because it's the natural way of flying for them (especially if you're a full scale pilot like myself too). So I find myself flying in manual and crashed my tip stall happy cularis 3 times because it tip stalled (anything below 25kph is risky on that airframe) because I just couldn't quite tell from the ground how slow I got when I was coming for landing and doing turns from downwind to base to final (low altitude, low speed, stall spin, and that's how real pilots also crash and die but at least they have the airspeed indicator in the plane they were supposed to watch).

magicrub, 2015-04-06:
Im just reluctant to add yet another flight mode. It would be nice to get
some comments from others on this.

marcmerlin, 2015-04-06:
@magicrub I totally understand not wanting to add a new flight mode. 
I was first going to suggest MANUAL_PREVENT_STALL 
That way, 
1) no extra flight mode
2) off by default, so it won't "you broke manual, it's supposed to be manual"

But it's probably a bad idea because if there is a bug, you won't be able to turn it off to a real manual.
So, I'd prefer MANUAL_PREVENT_STALL on to of MANUAL, but indeed the safest thing would indeed be to have a separate mode so that you can turn back to real MANUAL to turn off staff prevention if needed.
Not trying to convince you now, just thinking aloud.
Thanks for considering and asking others.

tridge, 2015-04-06:
what about we add it to TRAINING mode? Training mode is the same as manual until you hit roll/pitch limits. If you set large roll/pitch limits then it will feel like manual until you try to do a loop or roll.
We could auto-adjust the roll/pitch limits in training mode to take account of airspeed and the stall limits.

marcmerlin, 2015-04-06:
@tridge that's a great suggestion, this would work for me.I do roll planes and do loops, but don't need to do this on my glider, or can switch to manual for this, and then back to training mode for stall safety.

magicrub, 2015-04-06:
Good ole @Tridge to save the day!

tridge, 2015-08-14:
TRAINING mode now obeys stall prevention roll limits in master

closed 2015-08-14 by tridge

iskess, 2015-08-14:
Tridge,
Just for my knowledge, were any changes made in pitch, or just roll?

marcmerlin, 2015-08-30:
@iskess I believe stall prevention kicks in for both pitch and roll as required, so what @tridge added to training mode should be the same stall prevention code that is already there for FBWA and friends.