ArduPilot/ardupilot #2016: AP_Compass Compile Error

SCUBA82, 2015-03-25:
Trying to compile "Antenna Tracker" for my APM 2.5 in Arduino IDE . Since the new AP_COMPASS lib has been implemented I'm getting following compile error:

>C:\Users\xxx\Desktop\Arducopter\ardupilot-master\libraries\AP_Compass\Compass.cpp: In member function 'float Compass::calculate_heading(const Matrix3f&) const':
C:\Users\xxx\Desktop\Arducopter\ardupilot-master\libraries\AP_Compass\Compass.cpp:526: error: unable to find a register to spill in class 'POINTER_REGS'
C:\Users\xxx\Desktop\Arducopter\ardupilot-master\libraries\AP_Compass\Compass.cpp:526: error: this is the insn:
(insn 14 13 15 2 C:\Users\xxx\Desktop\Arducopter\ardupilot-master\libraries\AP_Compass\Compass.cpp:502 (set (reg:SF 48 [ D.9408 ])
        (mem/s:SF (post_inc:HI (reg:HI 16 r16 [56])) [9 <variable>.field.y+0 S4 A8])) 11 {*movsf} (expr_list:REG_INC (reg:HI 16 r16 [56])
        (nil)))
C:\Users\xxx\Desktop\Arducopter\ardupilot-master\libraries\AP_Compass\Compass.cpp:526: confused by earlier errors, bailing out

There seems to be an issue with the "headY" var in line 502. but I've no clou why. Tried to replace the "headY calculation" with a static float which compiles just fine.
Hope the Antenna Tracker is still supported on APM Boards?

madhunm, 2015-03-31:
Slight difference in the line numbers in the error message; however, same effect:

<code>
C:\Users\XXX\Documents\Arduino\ardupilot\libraries\AP_Compass\/Compass.h: In constructor 'Compass::Compass()':
C:\Users\XXX\Documents\Arduino\ardupilot\libraries\AP_Compass\/Compass.h:292: warning: 'Compass::_thr_or_curr' will be initialized after
C:\Users\XXX\Documents\Arduino\ardupilot\libraries\AP_Compass\/Compass.h:261: warning:   'uint8_t Compass::_backend_count'
C:\Users\XXX\Documents\Arduino\ardupilot\libraries\AP_Compass\Compass.cpp:277: warning:   when initialized here
C:\Users\XXX\Documents\Arduino\ardupilot\libraries\AP_Compass\Compass.cpp: In member function 'float Compass::calculate_heading(const Matrix3f&) const':
C:\Users\XXX\Documents\Arduino\ardupilot\libraries\AP_Compass\Compass.cpp:522: error: unable to find a register to spill in class 'POINTER_REGS'
C:\Users\XXX\Documents\Arduino\ardupilot\libraries\AP_Compass\Compass.cpp:522: error: this is the insn:
(insn 14 13 15 2 C:\Users\XXX\Documents\Arduino\ardupilot\libraries\AP_Compass\Compass.cpp:502 (set (reg:SF 48 [ D.9408 ])
        (mem/s:SF (post_inc:HI (reg:HI 16 r16 [56])) [9 <variable>.field.y+0 S4 A8])) 11 {*movsf} (expr_list:REG_INC (reg:HI 16 r16 [56])
        (nil)))
C:\Users\XXX\Documents\Arduino\ardupilot\libraries\AP_Compass\Compass.cpp:522: confused by earlier errors, bailing out</code>

SCUBA82, 2015-04-07:
It seems as if thats rather a compiler bug in avr-gcc 4.3.2 (arduino-1.0.3 ) than a bug in the AP_Compass library. 
I've just compiled "Atennatracker" with avr-gcc 4.8.1 (copied from arduino-1.6.3 to the ardupilot modified arduino IDE) without any error.
I've read about  compiler flags (-fno-caller-saves) which might also help to overcome that problem but there seems no way to pass additional flags to the compiler within arduino IDE.

magicrub, 2015-04-07:
Eventually we'll be setting warnings as errors in Travis so at least in gcc
we will catch any funny behavior. Not entirely sure how that would help for
avr-gcc but I hope it does.

rmackay9, 2015-04-08:
I've compiled antenna tracker with the modified version of arduino successfully so closing this issue.
http://dev.ardupilot.com/wiki/building-the-code/building-ardupilot-with-arduino-windows/

closed 2015-04-08 by rmackay9