ArduPilot/ardupilot #2022: Copter: compensate for changing air pressure during longer flights

hsteinhaus, 2015-03-29:
Our current altitude control logic for AltHold, Loiter, PosHold and Auto only relies on baro readings. The air pressure can change quite rapidly, espcially when thermal activity is present. This can easily result in an altitude error of > 40m for an one hour mission.   The resulting problems are obvious, from inadvertent entry into forbidden airspace to crashes during Auto-Land/RTL.

I would like to suggest adding a slow compensation for changing air pressure by adding small offsets obtained obtained from GPS altitude, after heavy low pass filtering and glitch rejection. This offset correction should be limited to phases of flight with constant altitude (e.g. Auto or Loiter sections without changing altitude demands).

The attached log shows the effect for a rather short section of flight. An error of almost 10 meters was present after only 20 minutes of constant-altitude Auto flight. If an RTL would have happened at this time, it probably would have descended into ground without slowing down early enough.

[image]

magicrub, 2015-04-05:
There's been a few threads talking about this, perhaps we should bring back ALT_MIX? It GPS alt is horrible for countries without SBAS such as Austrailia but otherwise GPS alt isn't bad enough to remove the feature. Tridge has a strong opinion on this, please chime in.
http://plane.ardupilot.com/wiki/arduplane-parameters/#gps_to_baro_mix_arduplanealt_mix

hsteinhaus, 2015-04-05:
I don't think that the GPS alt should be directly used for controlling the height of the vehicle, as the ALT_MIX featere did as far as I understand. 

I am currently working on a patch to use a filtered and averaged GPS alt signal to re-adjust the baro slowly during flight segments with constant alt target. First flight testing was successful - will send a PR as soon as I am sure that I have covered all corner cases.

rrr6399, 2015-09-09:
One thing I did recently was to use the barometric altitude of a ground-based Pixhawk connected to my ground station to add an altitude correction to the copter's altitude. (Just added a delta in altitude to the target waypoint in guided mode.)  This approach worked seemed to work OK, except that when it landed it thought it was much higher than it really was and landed hard. 

I really want to just change the reference altitude or pressure for the barometer in flight based on the ground based barometer. I noticed that there is a parameter called GND_ALT_OFFSET that just adds delta to the barometric altitude. If the ground based barometer is reading 10 m above ground level after a certain time due to changing weather conditions, I imagine I  need to add -10 m to the copter's barometric altitude. 

For copter, it appears to just add a delta to the barometric altitude where plane seems to have a tighter integration. I haven't tested this approach in the field yet to see if the copter has it fully integrated: i.e. if I set the GND_ALT_OFFSET to -10 m will the copter increase its altitude 10 m to hit the waypoint altitude? 

Has anybody else tried using GND_ALT_OFFSET to correct the barometric pressure of a flying copter to correct the indicated altitude based on a ground-based barometer?

magicrub, 2015-09-09:
You're using the right one, Randy. Plane has an additional one that offsets
everything, including GPS and baro and rangefinder which is just plain
wrong. I just recently fixed gnd_alt_offset. Just a few weeks ago it did
nothing at all!

rmackay9, 2015-09-09:
That param should be adjusted slowly though (like 1m at a time) or the EKF may freak out and stop using the baro for a bit.

magicrub, 2015-09-09:
.. Just like this...
https://github.com/diydrones/ardupilot/pull/2635

hsteinhaus, 2015-09-22:
As mentioned initially, I implemented an air pressure correction by using a heavily low-pass filtered GPS altitude ( https://github.com/hsteinhaus/ardupilot/tree/gps_qnh3 ) . I am flying this code regularly (with filter slightly adjusted to the even slower side) without any side effects. 

After looking through a few dozens of logs, I found an interesting detail: surprisingly the air pressure change seems to be heavily biased to the rising direction, no matter what time of day or what type of weather I am flying in. So I guess a (major) part of the effect is probably not related to true air pressure changes, but e.g. some thermal effects instead.

dab150, 2015-09-22:
Have you been performing a pre-flight calibration after the autopilot warms up? I spent some time trying various approaches to compensate for barometer drift and I noticed that the vast majority of the error comes from temperature changes in the first 10 minutes as the autopilot heats up. If I powered on the pixhawk and immediately armed and flew I would see a rise in my altitude of up to 8-10 meters. Whereas if I allowed the autopilot to warm up and performed a pre-flight calibration to reset my ground pressure, I wouldn't see nearly such an extreme change in altitude readings.

I think this happens because the pixhawk powers on and gets armed and the baro starts reading while it is at normal outside temperature. It will read a ground pressure that is temperature compensated correctly because the autopilot is at outside air temperature. However, as the autopilot heats up the baro begins compensating based off of the temperature inside of the autopilot, which can be much higher than outside temperature. Therefore you will get much different readings at the same altitude due to the baro reading much different temperatures.

If you perform a pre-flight calibration it will cause the autopilot to set a new gnd_pressure, one that is compensated with a much higher temperature. This means that the reading itself will be less accurate in terms of the actual outside pressure because the temperature isn't the actual outside temperature, but it will be more accurate relative to the readings you will get during flight, which is really what matters for accurate altitude hold, landing, etc.

This doesn't fix issues with changing weather and pressure changes, but like I said I think those issues are far less of an impact compared to this one.

Jman841, 2015-09-22:
I think this is inline with Trdige's tests with an external baro having far less drift. The external baro is much less susceptible to heat changes and only has to deal with actual weather and pressure changes.

hsteinhaus, 2015-09-22:
@dab150 
No, I didn't, but I don't have much hope that it would help. As you can seen in the picture of my initial post, the drift effect seems to be very persistent - no initial altitude change in a couple of minutes that could be preflight_calibrated away. Instead, the measured altitude changes slowly and persistently over the g the more than 30 minutes of flight (survey pattern in Auto mode). In other missions (up to one hour), there is the same picture.

WickedShell, 2016-02-09:
EKF2 can handle this by using the GPS as the primary altitude source (EK2_ALT_SOURCE = 2), so I'm closing this for now. It can be reopened if that doesn't sufficiently address the problem..

rmackay9, 2016-02-09:
TBH, I'm uncertain that using the GPS as the source is the best solution.  I would have thought that enhancing the EKF to use both sources would be the ideal solution (as suggested above).  Personally I've only ever seen this problem on a plane and it resulted in the plane being lost on a mountain because the error was about 70m after a 1300m climb.

hsteinhaus, 2016-02-09:
I am also not sure if the GPS alt really should be used for flight control. My approach was to slowly adjust the "ground pressure" by the observed difference between GPS alt and baro alt (https://github.com/hsteinhaus/ardupilot/tree/gps_qnh3)

This approach works quite well if a rigid low pass filter (cutoff frequency in the 1/120...1/300 Hz range) is applied on the delta. Using GPS to correct faster only adds noise to the altitude target IMHO.

gelazanskas, 2016-02-09:
@hsteinhaus lets think about what would happen in this situation:
The plane flies at high altitude for long enough to slowly adjust the ground pressure. Since we don't actually know what is the actual ground pressure (I assume it is calculated using some average value, say 1.2 kPa for every 100 meters), there is an error in the ground pressure value! If the plane descends rapidly and tries to land, the very slow low pass filter will be unable to compensate the error and the plane will hit the ground or flare early (in case of no sonar).

I assume in real planes the local tower broadcasts the ground pressure so in that case it is logical to input the ground pressure and then calculate the planes height.

Correct me if I am wrong :)

hsteinhaus, 2016-02-09:
@gelazanskas 
My hack adjusts the ground pressure for the difference between measured barometric altitude and measured GPS altitude, not for the absolute altitude above takeoff.

Hypothetic example: flying at 5000m AMSL (GPS) and  measuring 5005m AMSL (barometric). Conclusion: slowly readjust the ground pressure (it was measured during first arm) by an equivalent pressure of -5m (not 5000 something).

WickedShell, 2016-02-09:
The EK2_ALT_SOURCE uses the GPS as the primary, but it is still using the baro and mixing between the two. I don't know to what level but I do have some logs where noise on the baro (due to a weird/extreme aerodynamic change) caused the EKF to report itself as being unhealthy. I don't know to what extent the mixing is done, but it is mixing them :)

priseborough, 2016-02-10:
That's not correct. When EK2_ALT_SOURCE is set to 2 it will use the GPS 100%. While it is doing this, it constantly calculates an offset for the baro alt, so if GPS is lost it can switch to baro without a height jump.

hsteinhaus, 2016-02-10:
@priseborough Do you see any chance to do some kind of slow (filtered) barometer compensation directly the EKF? I think some kind of baro drift compensation is extremely important in a world with regulatory height limits. 

If you want to stay legal here, you have to make sure to stay below 100m for the whole flight. When assuming a possible drift of 15-20m per hour (my observations), you have to give up nearly 20% of the usable airspace just for allowing this drift for a one hour flight.

priseborough, 2016-02-10:
I think this is possible and it could be handled either in the EKF or in the Baro library. I don't think it warrant an additional state in the main EKF and could be as simple as a first order filter with rate limiter. 

I am working on equations to do baro and GPS blending using a range of estimators from single to four states. The error mechanisms that come to mind are:

GPS height wander - bounded with the limits indicated by some multiple of the EPV value
GPS height glitches - fairly random and really have to be handled using innovation consistency checks
Baro thermal drift - bounded and we have baro temperature information
Baro position error due to airspeed and attitude changes - bounded but very unpredictable
Atmospheric pressure drift - should occur at a steady rate
Pressure fluctuations due to local terrain and wind interaction (eg pressure is higher just upwind of a tall building) - very unpredictable but bounded

WickedShell, 2016-02-11:
Well since it sounds like I was totally incorrect I'll go ahead and open this again.

amilcarlucas, 2017-08-25:
AFAIK this is now in master and can be closed

OXINARF, 2017-08-25:
This is done in the EKF, but turned off by default and needs plenty of testing - I wouldn't close this just yet.

hsteinhaus, 2017-11-29:
@amilcarlucas, @OXINARF I guess this refers to the OGN_HGT_MASK param? Will give it a try as soon as weather permits a 1h mission and report back (OGN_HGT_MASK set to 5 as I do not have a range finder yet).

OXINARF, 2017-12-02:
@hsteinhaus Yes, that would be correct. Let us know how it went.

hsteinhaus, 2017-12-20:
Today I did a 1h mission with OGN_HGT_MASK=5. The mission was planned for 20m and no obvious deviation was observed. The baro alt drifted to +2, -4 and +3m again during flight, without significantly affecting the observed GPS altitude (without EK2_ALT_SOURCE set to baro). I think we can consider this problem as fixed.

closed 2017-12-20 by hsteinhaus

ddomit, 2024-06-05:
I have been doing tests with arducopter SITL with sim_baro_drift set to 0.015 and EK3_OGN_HGT_MASK set to 5, the drone does not correct for the baro drift, it starts losing height.

Any idea why this happens?

ddomit, 2024-06-05:
fsd