ArduPilot/ardupilot #2024: make throttle slewrates be uint16

tridge, 2015-03-30:
this will help with takeoff throttle slew for catapult launch

magicrub, 2015-04-05:
how does this help?

Naterater, 2019-01-05:
Is this the reason behind the minimum effective slew description here: http://ardupilot.org/plane/docs/parameters.html#thr-slewrate-throttle-slew-rate?

madmaverickminion, 2023-01-03:
Hi, is this issue still open?
I would like to work upon this.

Mac16661, 2023-09-09:
hello, I would like to fix this issue if it is still open.

TharakaUJ, 2026-03-01:
Hello! I’d like to work on this issue and put together a PR.

Before starting, I wanted to clarify a couple of things:

1. Is the goal primarily to increase the resolution of THR_SLEWRATE, or to increase both the resolution and the allowable range? I’m asking because in the previous PR there was discussion about potentially converting this to AP_Float instead of simply widening the integer type. 

2. As mentioned in the earlier discussion, changing the parameter width would need a  parameter conversion for existing configs. But that depends on the choice of number 1 above. Shall I come up with the conversion once we decide that.

Thanks!