ArduPilot/ardupilot #2026: Copter: Dangerous flights after EKF change

vmayoral, 2015-03-30:
Following from #1989 we changed to EKF but have been experiencing similar behaviors:

>when taking off in ALT HOLD the copter suddenly gets height really aggressively:

[image]

Log available [here](https://github.com/erlerobot/logs/blob/13ba4dd5c7476665b8170c42b0efa74e157242a5/pxf/copter/30-3-2015/69.BIN).

Our guess so far is that the Kalman filter parameters are wrong or is not properly tuned.

vmayoral, 2015-03-30:
@R-Lefebvre, @rmackay9, @tridge not sure how to diagnose this problem.  Any thoughts on this matter are welcome.

jschall, 2015-03-30:
Paul and I have patches that need to be pulled in to fix this sort of thing.

magicrub, 2015-03-30:
It may be worthwhile to note that #2010
<https://github.com/diydrones/ardupilot/issues/2010> was a recent bug that
@Tridge just fixed that was causing the EKF to not initialize properly and
therefor were reverting to DCM.

vmayoral, 2015-03-30:
@jschall and @magicrub thanks for the comments.

We are flying with our code rebased on top of the latest commits. @jschall could you point out which are those commits?

Regards,

vmayoral, 2015-03-31:
The problem seemed to be vibration-related due to the frame we were using.

vmayoral, 2015-03-31:
**Update**: The problem does not seem to be vibration-related.

[Here](https://github.com/erlerobot/logs/blob/a9498522bf2336ce0e9c105232313368f2cae2f5/pxf/copter/31-3-2015/83.BIN) is a flight with 2 imus enabled (MPU6000 and MPU9250) where the copter, as soon as it changes to `ALT HOLD` mode gets height aggresively. Can anyone make any sense of it?

(this seems to be a continuation of #1989)

rmackay9, 2015-03-31:
This is really a duplicate of an existing issue so closing.
https://github.com/diydrones/ardupilot/issues/1009

closed 2015-03-31 by rmackay9