ArduPilot/ardupilot #2027: Checklist logging

TerryPts, 2015-03-30:
It would be very useful for the controller to store a checklist of start-up data (arming checks etc) at the beginning of any flight which could be uploaded to Mission Planner or Tower and added to a more comprehensive manual checklist (user defined) and be easily printed. I fly commercially and a good paper trail is important. This would, at least partially, automate the process.

magicrub, 2015-04-05:
This is a good idea and the FAA is propose similar paper-trail requirements as well. Can you elaborate a little more on a possible work-flow?

peterbarker, 2020-06-18:
No more information has been forthcoming on this.  Workflows are important, but it's unclear what information the flight controller would be providing here.

closed 2020-06-18 by peterbarker