ArduPilot/ardupilot #2033: Pixhawk and EMAX 4in1 ESC issue

tonypascal, 2015-03-31:
Hello,
There seem to be a big problem with the Pixhawk and 4in1 ESC as reported here:
http://ardupilot.com/forum/viewtopic.php?f=21&t=5784&start=30
How could we deal with this problem, we are a handful of people not being able to fly our drones because of that...
Thank
Tony

R-Lefebvre, 2015-03-31:
This seems like a hardware issue to me, nothing we can do in software. As a couple people have indicated, the input impedence on the Emax ESC is not high enough, so the PWM pulse amplitude seen by the Emax ESC is not high enough. This is made worse on the Pixhawk , since it's driving the PWM at 3.3V instead of 5V as the APM2.5 does.

This is not the first time that an ESC has had this problem, there are a number of them out there that have it. Unfortunately, if this is the problem, there is really nothing we can do in software.

I suspect the fact that it sort of works maybe kind of sometimes, is an indication that you are on the ragged edge, and the PWM pulse is going over 3.0V, but takes a bit of time to get there, thus there's this off-set seen between min throttle.

Now, I have to ask, do you have a ground wire between the ESC and the Pixhawk?

This is the part that's not totally clear to me at this point.

I do not think the kinda/sorta/sometimes it works situation, is an indication that there is a problem with the software, as if there were, then all ESC's would have the same problem, not just a few.

CraigElder, 2015-04-01:
Hello all,

I took the problem the other way. I asked EMAX why it doesn't work. I wrote them an email and here is their answer:

"The current 4 in 1 ESC 25A doesn't support Pixhawk, we had noticed that and are currently modifying the firmware and doing test so that the new firmware will be compatible with Pixhawk flight controller. We will finish test in around a week. I will keep you posted about it."

So, hopefully, we should have some news next week.

Tony

closed 2015-04-01 by CraigElder

alduxvm, 2015-07-19:
Any update on this subject? or just the official response is "don't use EMAX with pixhawk"?

nickn7, 2015-07-19:
Ditto! I liked to use a Emax 4in1 on a quad but am
stuck using a Naza-Lite(V2) instead. Wish Emax would
get with the program. Won't buy another Emax ESC.

rmackay9, 2015-07-20:
It's tough for me at least to fix this very low level behaviour on the pixhawk.  The pins are pulled high when the pixhawk starts up rather than low like they did on the APM.

alduxvm, 2015-07-28:
Everyone that is still interested in the EMAX / Pixhawk dilema... Developers were correct. It was a ESC hardware issue. 

Unsolder a remove the two resistors marked in the picture and the ESC will work with pixhawk.

[image]

rmackay9, 2015-07-28:
Well, that is great news that the issue is understood.

garethbradley, 2015-09-12:
Hi,
Anyone know how to modify a EMAX BLHeli 20A ESC to work with the Pixhawk? Looks different to the above image.

[image]

Thanks!

alduxvm, 2015-09-12:
My suggestion will be to contact emax support... they might take some time in answering you, but they will tell you exactly which components to remove.

garethbradley, 2015-09-18:
I asked the question - no response yet...

drfailov, 2016-11-06:
Just if somebody has ESC wiht the same layout as mine, remove these two components to fix compatibility issues:
[image]

I don't know what exactly model is it, but found this solution by testing board pins.
Now it works fine.

magicrub, 2016-11-07:
what do those two caps do? What signals are on them?

magicrub, 2016-11-07:
This may be a good thing to add to the wiki. Any volunteers?

drfailov, 2016-11-07:
As mentioned in some thread in Diydrones, ESC don't work because it stuck in the bootloader while booting up. The reason of that can be different output signal from Pixhawk than the APM. 
So, in my opinion this 2 elements do some filtering to the signal, which already did in Pixhawk.

Steadyspike, 2017-10-01:
I have same problem with Emax Lightning 30A BLHeli and Pixhawk. Does someone know which components should I replace/modify? Find attached a photo.
[image]
[image]

drfailov, 2017-10-01:
Steadyspike, try to discover resistors between signal and ground and dismount it. I did my modification in same way.
Notice, that after that modification ESC will act randomly when not connected to FC.

Steadyspike, 2017-10-03:
It's no so easy... anyone with the same problem has fixed it?