ArduPilot/ardupilot #2039: i think therer is a bug in  both AP_Math/vector3.cpp and vector2.cpp

cat888, 2015-04-02:
vector3.cpp:
template <typename T>
bool Vector3<T>::operator !=(const Vector3<T> &v) const
{
    return (x!=v.x && y!=v.y && z!=v.z);
}

should be: return (x!=v.x || y!=v.y || z!=v.z);

vector2.cpp:
template <typename T>
bool Vector2<T>::operator !=(const Vector2<T> &v) const
{
    return (x!=v.x && y!=v.y);
}

should be: return (x!=v.x || y!=v.y);

magicrub, 2015-04-02:
> vector3.cpp:
> template
> bool Vector3::operator !=(const Vector3 &v) const
> {
> return (x!=v.x && y!=v.y && z!=v.z);
> }
>
> should be: return (x!=v.x || y!=v.y || z!=v.z);
>
> vector2.cpp:
> template
> bool Vector2::operator !=(const Vector2 &v) const
> {
> return (x!=v.x && y!=v.y);
> }
>
> should be: return (x!=v.x || y!=v.y);

nice catch!

tridge, 2015-04-02:
fixed, thanks!!

closed 2015-04-02 by tridge

magicrub, 2015-04-02:
Now the question is what, of anything, has been using those operands?
Surely the behavior would change.

tridge, 2015-04-02:
I've checked what was using it. Nothing was using the Vector2 != operator. There was one user of the Vector3 != operator, in AP_Param::set_and_save(), but the result wasn't used for vector parameters, so it didn't change any behaviour
Still worth fixing though!

magicrub, 2015-04-02:
totally worth fixing, thanks for the quick merge!