ArduPilot/ardupilot #2063: Compile error in APM2

jinchengde, 2015-04-08:
it looks like some error when compile code in Arduino-1.0.3

D:\Backup\Git\ardupilot\libraries\AP_InertialSensor\AP_InertialSensor.cpp: In member function 'bool AP_InertialSensor::calibrate_accel(AP_InertialSensor_UserInteract*, float&, float&)':
D:\Backup\Git\ardupilot\libraries\AP_InertialSensor\AP_InertialSensor.cpp:533:29: internal compiler error: Segmentation fault

magicrub, 2015-04-08:
This is not a code error, is a problem with gcc 4.6. We were getting that
problem with pixhawk so we upgraded to gcc 4.8.

What IDE and compiler are you using?

rmackay9, 2015-04-08:
I've closed this based on the discussion above.

closed 2015-04-08 by rmackay9

jinchengde, 2015-04-08:
@magicrub
I used gcc 4.7.2 and after upgrade to 4.8.2, everything is ok.
thanks for you help.