ArduPilot/ardupilot #2065: Add Tridge's geofence code to Copter

jabelone, 2015-04-08:
I was wondering if some kind soul would be able to add Tridge's geofence code to Copter?  Now that support for the APM2.x boards have been discontinued, what's stopping it?  (I would try, but sadly I I'm not good enough yet ;)  Thanks!

jmachuca77, 2015-04-08:
Hi, what geofence code are you referring to? Geofence is already supported in copter.

jabelone, 2015-04-08:
This one I believe - it supports a user definable polygon instead of just a circle with a set radius and height.  https://github.com/diydrones/ardupilot/blob/master/ArduPlane/geofence.pde

rmackay9, 2015-04-08:
Duplicate of this other issue so I'll close this one: https://github.com/diydrones/ardupilot/issues/1383

closed 2015-04-08 by rmackay9