ArduPilot/ardupilot #2068: Copter: takes too long to detect landing and trying to disarm causes flipping

andyp1per, 2015-04-08:
It takes a really long time to determine that it's landed. I have broken countless props, trying to disarm when landed only to have the thing flip over because it thinks I am applying yaw. The most reliable thing I can do is switch to stabilize, throttle at zero and just wait for it to automatically disarm. To my mind yaw applied at zero throttle should have NO effect on the prop speed. It should only be looked at as a disarming action. Right now there is effectively no way to disarm at all while the copter thinks it hasn't landed (which is often). This seems worse with 3.2.1 than 3.2.

R-Lefebvre, 2015-04-08:
Yaw input with zero throttle in Stabilize mode will not yaw the copter.  Maybe you knew that, but it wasn't really clear.

I think it might be possible to add the same functionality to Alt Hold and Loiter mode.  Randy, what do you think?

I can't see how it would be so horrible if you can't yaw while descending at full rate in Alt Hold.

andyp1per, 2015-04-08:
I'll try and find a log that shows this. I'm pretty sure that I have had flip overs because of trying to disarm through the zero throttle/yaw hard left routine in stabilize whilst on the ground. I know it shouldn't happen - but that's not what I am experiencing.

rmackay9, 2015-04-08:
There can be multiple causes of bad landing detection so when we have a dataflash log we can know better which of the possible causes is the actual issue.
AC3.3 (pixhawk only) has an improved landing detector so you may find it works better.

andyp1per, 2015-04-09:
The log is at https://dl.dropboxusercontent.com/u/108802517/90.BIN
The flip over is at the end. You can see me push the yaw over to try and get the thing to stop. I broke two propellers. Interestingly it doesn't register a crash which it has in the past.
(Any insight into the EKF_CHECK errors also appreciated! The copter flew fine. I am currently dubious that EKF is doing the right thing).

Wizardofoddz, 2015-04-12:
This is beyond annoying its a safety hazard. Destroyed a gopro because the machine would not turn off after it landed. 
http://www.droneshare.com/mission/83523/plot

R-Lefebvre, 2015-04-13:
Can you please put that log somewhere I can view it?

Wizardofoddz, 2015-04-13:
This link should work: 
https://api.droneshare.com/api/v1/mission/83523/messages.tlog

R-Lefebvre, 2015-04-13:
Ok, I've had a look, and this is what I see.

I guess you were doing some tablet flying first?  It appears an RCTx was connected, but the throttle stick was on the bottom.  This is a dangerous habit to be in.  The throttle stick should be raised to mid if tablet flying with an RCTx connected.

Then, it appears it landed successfully and switched to Loiter mode.  Then it appears it's being flown by the RCTx. It seems like it touched down, and tipped over.  It tipped to an angle of about 40-46 degrees (variable).  This is why the crash detector did not trigger.

Then, the mode remains in Loiter, then Land.  However, due to vibrations due to propeller contact with ground, it thinks it is flying again.  It has an inertial altitude that is changing.  It will not disarm in Loiter if it thinks it is flying. 

It could have been shut down by switching to Stabilize, then disarming.

rmackay9, 2015-05-02:
Now that AC3.3 is out, can you give it a try?

andyp1per, 2015-05-02:
Now that I have a reliable compass, yes sure :)
Incidentally I experienced this yesterday again with 3.2.1. The thing was still trying to fly even though it was on the ground, the problem is that once you are on the ground two legs stop the thing moving and the FC fights against this and causes the flip. Fortunately I flipped to a flight mode that made it take off again so no props lost this time.

andyp1per, 2015-05-08:
I'm going to have to try a longer flight test when the weather gets better. A simple flight test does not seem discernably different on landing and the Quad appears more wobbly in flight, so I might wait for RC4 before I try this (I'm also seeing altitude changes in loiter).

andyp1per, 2015-05-10:
Ok, I did a build from master today (so rc3++) and took it for a longer flight test. I have to say it *is* better especially in loiter/althold which is what I care about. The props spin down pretty promptly on landing almost as soon as it hits the ground. So good work!

magicrub, 2015-05-10:
Great, thanks for testing on master!

NortonAutonomous, 2015-05-10:
Gents, I had exactly the same problem with my new Y6 and had destroyed countless propellers trying to figure out how to fix the issue while using AC3.2. Running AC3.3 has solved my landing problems, thank you!

rmackay9, 2015-05-11:
ok great, news, thanks!

closed 2015-05-11 by rmackay9