ArduPilot/ardupilot #2069: Copter: minimum prop rotation ignored after landing

andyp1per, 2015-04-08:
The min spin speed for the props only works on the first arming. When you land at zero throttle and rearm the props do not spin. This may be something to do with my Spektrum TX.

rmackay9, 2015-04-08:
I think we should treat this as a support issue until it's confirmed to be a software bug (which is unlikely).
Sorry, but the issues list turns into a support form unless we're a bit strict about this.

This could be related to the battery voltage being low, looking at the RCOU messages in the dataflash and confirming that they're lower on the second rearm.

Perhaps Rob Lefebvre could help dig into this a bit.

closed 2015-04-08 by rmackay9

R-Lefebvre, 2015-04-09:
Yeah, I'm pressure sure this isn't normal, so a log would help.  How should he get it to me?

andyp1per, 2015-04-09:
I can put one on dropbox. The reason I wonder whether it's to do with my spektrum was that when I tried out my channel mapping patch I didn't get props spinning when armed at all.

So just to confirm - the minimum spin speed should always be in effect? i.e. on arm, on land - stabilize/zero throttle, on arm? Good to know, but that's not what I am seeing.

andyp1per, 2015-04-09:
So I have a log, but I'm not sure it's worth sharing as it shows exactly what you would expect - RC3 IN at 1100 but RC1-4 OUT all consistently at 1200 whenever the throttle is put back down to zero. This is not what I am seeing though on the quad. The first time the props spin fairly aggressively the second few times they stop entirely. What would cause this? The battery is fresh and this happens as soon as I land (so only a couple of minutes after a fresh charge). Presumably I should run a test where I increase the minimum spin speed to see what happens?

R-Lefebvre, 2015-04-10:
Yes, there might be some hysteresis on the ESC input.  I'd try raising it 50-100, see if that makes it work properly. Obviously be careful if it's going to actually lift off, that would be bad as it won't be stablizing so it'll flip over.

andyp1per, 2015-04-13:
So I raised it 50, took off hovered for 30s or so and then landed. Props stayed spinning. Did the same again but for longer, props stayed spinning but slower. Did it again, hovered for a few minutes, landed and the props stopped. So it's related to how long it is in the air (and I guess possibly based on the battery voltage). How is the percentage calculated? Could it be something as simple as the spin speed calcuation is related to the battery voltage at the time and as it drops so does the spin speed?
Tried another experiment - hovered, landed - props stopped - waited for it to disarm, re-armed and the props started spinning again. So it definitely feels like whatever calculation is done on arming becomes out of date as the battery voltage drops.