ArduPilot/ardupilot #2070: Copter: Feature Request: Allow Advanced Use to permit PID values outside of constrained ranges in MP Extended Tuning

Hands0n, 2015-04-08:
I am raising this in reference to an issue I raised earlier and which is now closed, see https://github.com/diydrones/MissionPlanner/issues/831 

https://github.com/diydrones/MissionPlanner/issues/834 also refers.

I would like to suggest a feature in Extended Tuning that allows "Advanced Use" whereby when enabled the user is not constrained to ranges of values possible. The "Advanced Use" switch would require a deliberate action on the part of the person using Mission Planner. By default it would be "Off" (ie. constrained PID ranges).

Resorting to Full Parameters as a solution takes away from the convenience and functionality of MP's Extended Tuning panel.

Why? I maintain ArduCopter in five different aircraft and in at least three of them the constrained ranges are not suitable. Mission Planner, as a tool, is wonderfully comprehensive and flexible. Other panels allow Advanced Use, so I would appeal for this to be provided in a future version of MP.

rmackay9, 2015-04-08:
Sorry if you're getting the run-around here but all we can do in arducopter is increase the allowable range of the parameter values.  So for example, it looks like 0.25 is the maximum for the Roll/Pitch Rate P.  We could increase that to say 0.28.  We can't do all the other things mentioned like turn-off range checking min MP by making an ardupilot/copter change.

Hands0n, 2015-04-09:
Hi Randy,

I think the problem is Rate Roll/Pitch P and D have been set with a maximum that is too low.

So in my example, my working Rate Roll/Pitch P is 0.2630.  This has been arrived at by in-flight tuning using Channel 6 in MP prior to 1.3.23 (in my case I tuned my hex up in MP 1.3.19). 

If you were to set Rate Roll/Pitch P to 0.28 and D to 0.0060 that would work for us with this frame type.  

NB: These frames are an intentionally highly rigid aluminum hex and quad design, currently using 2212 920KV motors and 8045 props on 3S LiPo.

R-Lefebvre, 2015-04-09:
I've also found the PID range too small for most of my copters.  I just hesistate to increase them as I'm not sure if it's common, or just a few of us?

rmackay9, 2015-04-09:
Ok, I'll check with Leonard about 0.28 and if he agrees then we all agree and I'll make the change in master.
By the way, the max for RATE_RLL_D (and RATE_PIT_P) is 0.02.  Did you mean 0.06?

Hands0n, 2015-04-09:
Hi Randy, not a math genius at work here :-)   0.02 is absolutely fine for Rate_RLL_D.   

The incumbent maximum 0.2 RATE_RLL_P and RATE_PIT_P are the troublesome ones in this report, so if they were adjusted to 0.28 that should cover our experience (ie. where our optimum P is 0.236). 

Thank you
Danny

rmackay9, 2015-04-09:
ok, we've increased the param definition's max to 0.3 so hopefully we're good and I will close this issue.
https://github.com/diydrones/ardupilot/commit/70fbb3c67efa56944da128a1d6a015c2e263d51c

closed 2015-04-09 by rmackay9

Hands0n, 2015-04-09:
Thank you very much for this. I'm sure it will save a lot of grief :-)  Much appreciated.

Rgds
Danny