ArduPilot/ardupilot #2073: AP_NavEKF::SetFlightAndFusionModes makes bad onGround determination when flying with no GPS/OF

John-Ingersoll, 2015-04-09:
The main purpose of SetFlightAndFusionModes seems to be A) determine if the vehicle is on the ground and B) when it takes off, perform some initialization logic for the yaw and wind states.

This code is entirely disconnected from the ArduCopter arm/disarm, landed logic that has been worked so extensively.

The Application level should make the determination that the vehicle is on ground or not and tell the AP_NavEKF that it is through a set command.

WickedShell, 2017-02-10:
I believe this can be closed as I believe this was an EKF1 issue, at least I can't find the referred to function in the current code base, does the same issue still exist?

OXINARF, 2017-02-10:
Although this exact method doesn't exist anymore it is true that the EKF and the vehicles still do independent determination of grounded/in-air status. I'm not sure how important that is though.