ArduPilot/ardupilot #2087: Plane: Break LAND_APPROACH stage into two segments

magicrub, 2015-04-10:
To give better flexibility on landing sequencing, I propose we break up the flight_stage landing sequence into two segments: LAND_APPROACH_HIGH and LAND_APPROACH_LOW.

A new param would be used as an altitude as to when those would be divided. This would allow the last segment of a landing to be treated differently than the rest of the approach. I usually start my landing approach at 100-150m and only at the flare does it change behavior for a critical landing. It would be nice to treat [user defined height] as critical too or special in some way.

Helpful uses:
- Issue #2079 where the fence can be ignored during the last portion
- turn on landing lights or a siren or release a drag chute
- ignore RTL or other flight mode requests

iskess, 2015-04-10:
Please don't ignore flight mode requests. MANUAL should always be honored as a rule. 
I use TRAINER to take over because it provides me protection for over controlling in a critical situation. 
FBWA is a safe way for less experienced pilots to assume control smoothly. 
RTL is a commonly used method of GO-AROUND, at least until RC triggered GO-AROUND is implemented.

magicrub, 2015-04-10:
this feature could be disabled by setting the alt to zero (probably the default?) which would mean you're doing the _HIGH all the way down until you flare giving you the exact same behavior as today.

magicrub, 2015-09-12:
I like this idea for a pre-flare stage where the airspeed can be lowered just before flare, not 100m up in the air with TECS_LAND_ASPD. Perhaps we can do other things like turn off stall prevention or <fill in the blank with other things humans do to bleed off airspeed as we land>

iskess, 2015-09-12:
I don't use TECS_LAND_ASPD for just that reason. I want to maintain a higher stall margin an approach for wind shear, until I cross "the fence" (approach edge of the field). 
This proposal would allow me to start bleeding speed off earlier than the flare, while maintaining a safe stall margin up to that point. 
Great idea. Thank you.

magicrub, 2015-09-12:
For the sake of historical record, all proposed landing abort methods have been pushed to master and are available now. Includes throttle abort trigger which causes a pitch and climb out maneuver proceeded by a go around via command alt as need command after land, jump to do land start, or mission index minus one. In the process of adding the pitch and throttle up, a new flight stage was added called land abort.

iskess, 2015-09-12:
Tom,
Thanks for making this all happen!

magicrub, 2016-01-30:
fixed with https://github.com/diydrones/ardupilot/pull/3567

closed 2016-01-30 by magicrub