ArduPilot/ardupilot #2089: PMW LIDAR-Lite not working on Plane

zlite, 2015-04-11:
The PWM Lidar-Lite works on the current copter code, but not Plane (Both Philip and I tried). The cable is correct and it's plugged into the right PWM ports, and the Type is set to 5 etc. No joy....

magicrub, 2015-04-11:
I added a power down feature not too long ago, I'll double check that it
didn't break something

magicrub, 2015-04-11:
Chris,
It's working for me on master, can you give me more info:

What firmware version, master?
What are all your RNGFND values?
Do you have the stop pin installed and with the resistor? I presume so
since you just updated the wiki pic.

Does it [not] work on the bench? In the air? Auto? Manual?

My settings are:
RNGFND_FUNCTION,0
RNGFND_LANDING,2
RNGFND_MAX_CM,3000
RNGFND_MIN_CM,150
RNGFND_OFFSET,0
RNGFND_PIN,54
RNGFND_RMETRIC,0
RNGFND_SCALING,1
RNGFND_SETTLE,0
RNGFND_STOP_PIN,55
RNGFND_TYPE,5

-TomP

zlite, 2015-04-11:
Code is ArduPlane V3.3.0beta1 (55019abd) 

Stop pin installed with a 470 Ohm resistor

My settings are below. It looks like the important missing bit is RNGFND_PIN, 54, which is not mentioned in the documentation! When I'm back at my desk I'll give that a shot. 

[image]

zlite, 2015-04-11:
PS: I just added "RNGFND_PIN, 54" to the list of required param changes in the wiki.

magicrub, 2015-04-11:
Hi Chris,

RNGFND_PIN is required for PWM, so that would definitely make it not work
for PWM mode. Also, i see you have RNGFND_MAX_CM as 4900 which is 49
meters. That is beyond the spec of LidarLITE which is 40m. That would mean
if you are in the air at 100m it will max out at 40m and that would be
taken as a valid accurate range. That needs to be lowered down to something
under 4000. I have never seen above 3850 as a reading and so I've set mine
conservatively to 3000. Same with the MIN range, I found 20cm to be too
close so I have an overly conservative value of 150 (1.5m). When landing, I
should have flared by this point so it's ok in my application.

-TomP

magicrub, 2015-04-11:
ah! I just realized that my PR
https://github.com/diydrones/ardupilot/pull/2055 is still pending and is
not in master yet.

tridge, 2015-04-11:
Hi Chris and Tom,
The RNGFND_PIN parameter is not used at all if RNGFND_TYPE is set to 5 for PWM. 
The pin number is hard-coded inside the PX4 PWM input code (due to muxing restrictions)
Cheers, Tridge

zlite, 2015-04-11:
Ah, got it. I'll remove my addition from the Wiki and wait for the PR to be accepted. Thanks all!

magicrub, 2015-04-11:
Well the PR isn't a fix, it's an enhancement to keep it powered down until
landing approach due to GPS jamming issues. Plus it's only utilized
with RNGFND_LANDING=2
which is a new value

closed 2015-04-13 by CraigElder